thotro/arduino-dw1000 #183: Correct IDE version / library version / example?

Saflight, 2017-04-02:
Dear all,

I recently tried to make the ranging examples working. The assembly I have is exactly the same as https://sites.google.com/site/wayneholder/uwb-ranging-with-the-decawave-dwm1000---part-ii

I had my system working for 1 anchor 1 tag (RangingAnchor/Tag codes), but after having updated IDE version I can no longer step back my system for working condition. Could you please confirm the following for reliable results?:

-IDE version necessary?:

-Library version necessary?:

-Sample code necessary(DW1000Ranging_ANCHOR/TAG or RangingAnchor/Tag?:

Thank you in advance mates

prideofisland, 2017-04-02:
Hi there, 
The '1 Tag & 1 Anchor' system works with all Arduino IDE versions up 1.6.8 (I've tried almost all of them). In order to run 3 Anchors and 1 Tag you propably have to use 1.6.8 version. 
You don't need to change the code to launch the system. Try 'Basic connectivity' and tell us what you get in the serial monitor. Maybay it's a hardware problem.

az-z, 2017-04-02:
Just use pio.

Saflight, 2017-04-02:
Hi all,

Miraculously I got it working!!! But I have to say that I only got it working after replacing "INPUT" by "INPUT_PULLUP" in "DW1000.cpp" file (lines 120 and 259 for me). This was recommended solution on previous thread, however I've also read that doing this may not be good for DWM1000.

What is your feedback about that? Did you get it working without replacing those lines? Could you find any other alternative to make it work? I'm afraid of destroying my DWM1000 as I finally got them to work.

In the other hand and just for the rest of the users I'm using arduino IDE 1.6.9 and last version of the library. I got working the case of 1 anchor & 1 tag and also the case of 3 anchors & 1 tag.

Thank you in advance for your support

hu265, 2017-04-03:
Welp... Currently I'm using Arduino IDE v. 1.6.10 and v.0.6 release of the library. Unfortunately, the Wayne's board didn't work for me so I hooked up my parts with wires(for now). I got the RangingTag/Anchor to "work" for my 1 tag + 1 anchor case but I've been getting negative values for the distances and I'm not sure how to fix that(I was just gonna open another issue to ask about this but I feel bad for annoying everyone....) So yea, sorry if I'm not being very helpful here x(

Saflight, 2017-04-03:
Did you get no results when just pluggin your pro mini on wayne's holder?

Assembly might be pretty sensitive to connections, so wiring may decrease the system performance.

prideofisland, 2017-04-03:
> But I have to say that I only got it working after replacing "INPUT" by "INPUT_PULLUP" in "DW1000.cpp" file (lines 120 and 259 for me). This was recommended solution on previous thread, however I've also read that doing this may not be good for DWM1000.

I have in these lines only 'INPUT'. 
`		pinMode(_rst, INPUT);`
It works like a charm from my side.  I'm using arduino IDE 1.6.8  and newest release of the library. Do you use arduino Pro Mini 8Mhz? I've tried with Arduino Pro Micro but it didn't work and I switched to Pro mini. 

>  I'm afraid of destroying my DWM1000 as I finally got them to work.

Make backups in order to save your current progress ;)

> Unfortunately, the Wayne's board didn't work for me so I hooked up my parts with wires(for now).

I had also some problems with hardware at the beginning. Finally, I designed my own board with power supply. 

> I got the RangingTag/Anchor to "work" for my 1 tag + 1 anchor case but I've been getting negative values for the distances and I'm not sure how to fix

You established connection between modulus.  You got over the hump! Once I have received extremely high values (about few hundreds meters). I connected VDD3V3 pins with VDDAON and it fixed the problem.

Saflight, 2017-04-03:
@prideofisland 

I'm using Arduino Pro Mini 16MHz 5V (chinese one)

Of course I'm always doing backup but I meant something more like physical damage of the DWM1000 chip, is that possibly happening?

Which arduino board did you design your PCB to fit in? If I finally get everything working as it should I would probably like to give it a try with mega2560 or arduino due, any feeling on that?

prideofisland, 2017-04-04:
@Saflight 
From my experience, DW1000 is really forgiving of mistakes (either software or hardware). No need to worry! I used Arduino Pro Mini 8Mhz 3,3V.

Give a try and then share your observations with us. I've tried with Arduino Pro micro but only some of example programms worked. Well I wish you luck anyway.

hu265, 2017-04-06:
Hi @prideofisland 
Did you mean connecting the VDD3V3 and VDDAON pins on the DWM1000 chip?

prideofisland, 2017-04-06:
Hi @hu265 

Yep. I connected these pins as shown in the image.
[image]

closed 2018-08-30 by Rotzbua