thotro/arduino-dw1000 #185: Static Anchor Addresses

christoshadjiaslanis, 2017-04-12:
I am implementing 2D positioning using the TWR implemented in the DW1000Ranging_ANCHOR/DW1000Ranging_TAG. Does anyone know how to set static addresses for the Anchors because everytime I turn them on and off again their short address seems to change.

prideofisland, 2017-04-12:
Hi @christoshadjiaslanis , 
You are right. Anchors change their addresses. Check in DW1000Ranging.cpp  

```
_currentShortAddress[0] = random(0, 256);
_currentShortAddress[1] = random(0, 256);
```

christoshadjiaslanis, 2017-04-12:
@prideofisland thanks for the response. I implemented static addresses for anyone interested:

First change DW1000Ranging.h 

```
	static boolean addNetworkDevices(DW1000Device* device, boolean shortAddress);
```

Then change DW1000Ranging.cpp

```
void DW1000RangingClass::startAsAnchor(char address[], const byte mode[], const byte anchorAddress) {
	//save the address
	DW1000.convertToByte(address, _currentAddress);
	//write the address on the DW1000 chip
	DW1000.setEUI(address);
	Serial.print("device address: ");
	Serial.println(address);
	//we need to define a random short address:
	randomSeed(analogRead(0));
	_currentShortAddress[0] = anchorAddress;
	_currentShortAddress[1] = 0;
	
	//we configur the network for mac filtering
	//(device Address, network ID, frequency)
	DW1000Ranging.configureNetwork(_currentShortAddress[0]*256+_currentShortAddress[1], 0xDECA, mode);
	
	//general start:
	generalStart();
	
	//defined type as anchor
	_type = ANCHOR;
	
	Serial.println("### ANCHOR ###");
	
}
```

Then in the DW1000Ranging_Anchor.ino:

```
 DW1000Ranging.startAsAnchor("82:17:5B:D5:A9:9A:E2:9C", DW1000.MODE_LONGDATA_RANGE_ACCURACY, anchorID);
```

closed 2017-04-12 by christoshadjiaslanis

Rotzbua, 2017-04-26:
@christoshadjiaslanis please make a pull request thanks

duplicate  #79