thotro/arduino-dw1000 #186: Negative range with 3 anchors

christoshadjiaslanis, 2017-04-12:
I am implementing TWR with 3 anchors (DW1000Ranging_ANCHOR.ino). The ranges are accurate until the 3rd anchor is introduced at which point the anchor that connected 3rd gives negative distances.

Has anyone come across this issue and have a fix?

christoshadjiaslanis, 2017-04-12:
This issue seems to have gone after starting the anchors with non-random short addresses. I'm not going to close the issue in case the problem reemerges and this is just a coincidence.

prideofisland, 2017-04-15:
Hey @christoshadjiaslanis ,
I've seen negative values at the beginning of my work with DW1000. It occured when the distance between devices was small (under 1m). I can honestly say that DW1000 are not working properly in small range (under 2m).

Rotzbua, 2017-04-26:
duplicate #121

closed 2017-04-26 by Rotzbua