thotro/arduino-dw1000 #187: Multiple Anchors affect range

christoshadjiaslanis, 2017-04-13:
I have a setup with three anchors running DWM1000Ranging_ANCHOR and 1 tag running DWM1000Rangin_TAG. I have noticed that as more anchors are introduced the spatial accuracy drops dramatically (1 anchor has really good accuracy, 2 anchors have reduced accuracy and 3 is horrible). Has anyone come across this? 

Could it be interference? Should I try changing the channel?

prideofisland, 2017-04-15:
Hey @christoshadjiaslanis ,

I have noticed it few months ago also. As you probably know, Tag broadcasts POLL msg to all Anchors. Then anchors replay with POLL_ACK msg. This message is send in pre-determined order.  Anchor which is first in DeviceLists has other response time than the others. They cannot response simultaneously because tag needs some time to process the signal.  The delay supposed in this library:
Anchor1 - 7000 microseconds
Anchor2 - 21000 microseconds
Anchor3 - 35000 microseconds

Do you have a constant offset? You can try play with antennaDelayBytes value, substract offset before printing it or something else. I'm not sure that changing the channel will bring a lot of improvement. Check maybe various modes like: LONGDATA_FAST_ACCURACY or MODE_LONGDATA_RANGE_ACCURACY.

Best regards, 
Pride

christoshadjiaslanis, 2017-04-16:
@prideofisland thanks for getting back to me. I will look into it and let you know.

Rotzbua, 2017-04-26:
seems to be a duplicate of #50

closed 2017-04-26 by Rotzbua

Rotzbua, 2017-04-26:
closed in favour of #181

Titibo26, 2021-08-02:
HI #50 and #181 are different (its not about multiple tag and it's a precise issue here).
Did you find a solution for this problem ?