thotro/arduino-dw1000 #188: Maximum Range

Saflight, 2017-04-18:
Hi all,

I'm now facing a wall on my project as we are hardly getting any signal above 10m far. I've read from decawave that range could reach up to 300m, however me and my team are pretty far from that.

Any suggestion such as (code or hardware modification) in order to improve that? I'm currently working with DW1000Ranging_ANCHOR/TAG examples, have not touched nothing yet on libraries.

Please, I need at least 50m range, anyone has been able to reach 50m or higher ranges?

Many thanks to all for your cooperation

christoshadjiaslanis, 2017-04-19:
Are you working in LOS conditions?

Saflight, 2017-04-19:
Yes, absolutely

kuek, 2017-04-20:
Hi, you can change to a lower channel (i.e. 3.5 GHz and longer preamble)

Check out my hackaday post. 50m works

https://hackaday.io/project/18296-localino-open-source-indoor-localization-system

Saflight, 2017-04-20:
Hi kuek, thank you for your indications, however I'm not able to see how you changed those parameters I was mentionning. Could you please make any specific reference for that?

Great project by the way

kuek, 2017-04-20:
Thanks!
It is written in the DW datasheet.
Unfortunately i dont have the datasheet available right now, i would then look it up for you.

Saflight, 2017-04-21:
Hey,

For everyone listenig to this topic. I found all the modes under one of the files on dw1000-master library (in src folder of course). It should be possible to select any of the predefined modes shown there (6 modes) or manually change each of the parameters.

After playing for 5 mimutes with that I got range over 80m. I didn't verify yet the measurements accuracy but is already a good signal to have such a long range.

As per decawave support team they say that 160m should be easily achievable, so I will keep playing with those parameters and get back if any improvements are achieved.

As a note, I was using DW1000Ranging_ANCHOR/TAG codes with wayne's assembly.

Rotzbua, 2017-04-21:
@Saflight It would be grate if you can add your experience to the wiki. Thanks. https://github.com/thotro/arduino-dw1000/wiki

kuek, 2017-04-21:
perfect! glad you got it working! RF is pretty straight forward once you obey some rules... ;-)

skadiexe, 2018-02-13:
my results with DWM1000
channel 1, 110 kb/s, 2048 preamble lenght
range test LOS: 
226m maximum range i got, signal lost
195m signal achived
80m stable connect without loss of the packets
20m stable connect thru human body