thotro/arduino-dw1000 #191: Frame filtering

Diman142, 2017-04-20:
Hi all

I started to work with arduino+DW1000

I tryed examples, researshed DW1000.cpp library, and implemented few simple ideas.

Now I want to try frame filtering. I read user manual, ASP10 and ASP13 by Decawave.
Try to form MAC frame for frame filtering.
I use next configure on tranciever part.

`#define PAN_ID 0x5555
byte MAC[16] ={0x84, 0x22, 0x84, 0xAA, 0xAA, 0x88, 0x88, 0xAA, 0xAA, 0x44, 0x44, 0x01, 0x00, 0x00}; 
uint16_t Short_AD = 0x2222;

void setup() {

  DW1000.newConfiguration();
  DW1000.setDefaults();
  DW1000.setDeviceAddress(PAN_ID);
  DW1000.setNetworkId(Short_AD);
  DW1000.enableMode(DW1000.MODE_LONGDATA_RANGE_LOWPOWER);
  DW1000.commitConfiguration();

  transmitter();
}`

and transmit my MAC array using DW1000.setData(MAC); function 
On the reciever part I use next config:

#define PAN_ID = 0x5555;
uint16_t Short_AD = 0x1111;
void setup() {
  DW1000.newConfiguration();
  DW1000.setDefaults();
  DW1000.setNetworkId(PAN_ID);
  DW1000.setDeviceAddress(Short_AD);
  DW1000.setFrameFilter(true);
}

And reciever don't understand messages by tranciever 
I don't understand what wrong with my code maybe I forgot something or something else 
In mac header I point: Data frame and use PAN ID compression field bits. I don't use FCS bits and last 2 bits equal zero

Have you simplest example with frame filtering?

P.s It is part of code other configurations as in the example

Thank you for attention!)

Rotzbua, 2017-04-20:
@Diman142 Please use code highlighting with markdown https://guides.github.com/features/mastering-markdown/

Look at the old code https://github.com/thotro/arduino-dw1000/blob/master/src/DW1000.cpp#L1177 .
I disabled it by default because it caused some issues.
Best.

Diman142, 2017-04-20:
Ok I will use markdown I am sorry for first post.
But I in my code I Enabled frame filtering manually

`DW1000.setFrameFilter(true);`

I think that was enough because i do it after 
`setDefaults();`

Or you want to tell me about tranciever part.
Shold I enabling filtering on tranciever part?

Diman142, 2017-04-21:
Now I try to use DW1000Mac.cpp library
Is this library work?

Rotzbua, 2017-04-21:
From DW1000Mac.cpp:
> @todo everything, this class is only a prototype

So depends on what you need. It is limited, it's used by dw1000ranging.h.

Diman142, 2017-04-22:
I need to implement frame filtering and I want to do it using
`DW1000Mac::generateShortMACFrame` 
In my tranciever and
`DW1000Mac::decodeShortMACFrame`
in my reciever

Does this class work?