thotro/arduino-dw1000 #194: Rotating tag changes the measurement

reycreator17, 2017-04-28:
Hi everyone. I'm using dwm1000_Tag and _anchor. 1 anchor and 1 tag connected.

 I just want to ask, why is it that when I rotate either the anchor or tag, the measurement of my serial monitor (of my tag) changes about +- 50 cm?

Is it some sort of normal or am I the only one having this?

Thanks 

Rey

christoshadjiaslanis, 2017-04-28:
this is normal. The antenna is not isotropic. If the antennas are facing each other perfectly the most accurate and consistent ranging should be achieved.

reycreator17, 2017-04-29:
So much thanks @christoshadjiaslanis 

BTW, what does the 16384 in the `writeValueToBytes(antennaDelayBytes, 16384, LEN_STAMP);` really mean?

Should each anchor and tag have different values of them to calibrate the precision?

Rotzbua, 2017-04-29:
@reycreator17 Do not mix topics in issues.

closed 2017-04-29 by Rotzbua