thotro/arduino-dw1000 #195: How can we get distances between a tag and more than one anchor?

wangwangguoguo, 2017-05-02:
I've managed to get the distance between 1 tag and 1 anchor.
But how can we add a second anchor?
We still use the same DW1000Ranging_ANCHOR.ino and DW1000Ranging_TAG.ino, or should I change some lines of codes?

KeyYD, 2017-05-03:
The time line should be controlled in this. #181

wangwangguoguo, 2017-05-14:
That is to say, I needn't change anything if I just want to get the position of a tag by receiving 3 distances with different anchors? @KeyYD

KeyYD, 2017-05-17:
@wangwangguoguo Yes, but more precisely, if you want to use multi anchor, a certain address(not randomly) for every anchor will be better.

wangwangguoguo, 2017-05-17:
@KeyYD 
[image]
2 questions

1：
I've changed the code in DW1000.cpp
`volatile uint8_t      DW1000RangingClass::_networkDevicesNumber = 0;`
in DW1000.h
`static volatile uint8_t      _networkDevicesNumber;`

But as is shown in the picture the 3rd anchor can't get connected. Have I changed the code correctly?

2: 
The short address is random all the time although I define it like this:
`	_currentShortAddress[0] = 1;
	_currentShortAddress[1] = 0;`
[image]

I get exactly the same problem and I do as is suggested in #179  but the pb hasn't been solved. I can't find where is wrong. Do you have any idea?

PS: I've also assigned  a certain address  to every anchor differently.

closed 2017-05-22 by wangwangguoguo