thotro/arduino-dw1000 #199: Reduce library size

mattweb67, 2017-05-26:
Hello !

First of all I would like say thank you to all who participate to the development of this great library which work very well.
My issue is not really about the functional aspect of the library but about the size of it. I use the library with an ATMEGA328P and I also use a CAN library. My program is based on the DWM1000Ranging and the program is really too big. I dont know how I could reduce the size of my program by commenting or deleting some unused function of the library. Some ideas ?

Thanks

Matt

prideofisland, 2017-05-26:
Hi @mattweb67 ,
I have two version of library. One for tags, seconds for anchors.  In anchors version I deleted/commented out code for Tags and vice-versa. Try to avoid printf -they're really memory-consuming. I commented debug also (first text with device address & ID, channel etc,)

mattweb67, 2017-05-26:
Hi @prideofisland 
Good ideas I try that now, currently I use 110% of the flash memory of the ATMEGA :/

Thanks

Rotzbua, 2017-05-26:
@mattweb67 This is a general question. Please ask on https://stackoverflow.com/ or https://forum.arduino.cc/ . Thanks

closed 2017-05-26 by Rotzbua