thotro/arduino-dw1000 #201: basic send/receive

Paultn12, 2017-06-01:
Hi,

I am trying to get the basic send/receive to work. On the sender I seem to be managing to send, but on recieve this is all I get is:

### DW1000-arduino-receiver-test ###
DW1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 0A, Short Address: 06
Device mode: Data rate: 110 kb/s, PRF: 16 MHz, Preamble: 2048 symbols (code #4), Channel: #5

I have double checked over wiring and I am setup with the adaptor boards and a couple 3.3v arduino pro minis. Power to the arduinos is via 2 arduino unos (3.3v) and power to the dwm1000 is via an external energy supply to achieve the current.

on the sender end I get the above plus:

Transmitting packet ... #0
ARDUINO delay sent [ms] ... 28
Processed packet ... #0
Sent timestamp ... 9580229.00
DW1000 delta send time [ms] ... 9580.23
Transmitting packet ... #1
ARDUINO delay sent [ms] ... 26
Processed packet ... #1
Sent timestamp ... 9744079.00
DW1000 delta send time [ms] ... 163.85

I have had a look on these forums and others and cannot find the issue.

Thanks for looking.
Paul

Paultn12, 2017-06-01:
It seems like it could be an issue with the interrupt pin. Sometimes only sends when I have it connected and sometimes only if not. Not as much luck with the receive sketch

Rechecked wiring again and even changed DWM1000 - still nothing solid.

Some post seem to mention the antenna model - I have UWB66 if that is any use.

Rotzbua, 2017-06-02:
Please use the template next time. Thanks.
https://github.com/thotro/arduino-dw1000/blob/master/.github/ISSUE_TEMPLATE.md

closed 2017-06-02 by Rotzbua