thotro/arduino-dw1000 #202: How do I get the 3 range data by using 3 anchors and 1 tag?

HYUMell, 2017-06-01:
### Step 1: What do you want?
Hi, I want to get 3 range data because I need a robot tracking system. So, I use DWM1000 module and arduino pro mini, but actually I don't know UWB communication.
I only run example for DW100Ranging_ANCHOR and DW1000Ranging_TAG.
I got 3 range data from 3 ANCHOR. I want to use this 3 range data to calculate trilateration.
How can I change the code so that the range values from tag and anchor with 3 anchors and 1 tag?

* Report an issue.
* Improve code (and need help).
* Suggest a new feature.
* Present a new hardware adapter board or wiring.
* Want to report my experience or measurement. (Please also add them to the wiki!)

Maxai2, 2017-09-07:
Hi @HYUMell !
Can u tell me how u got 3 range from 3 ANCHOR? By Arduino IDE Monitor?
I also work with this system, and I don't know what I do wrong....

werty37, 2017-09-15:
I am also having the same query. Is this information available through just any single anchor which is connected in the same UWB network. Someone please clarify.

nagendra092918, 2019-08-22:
i am also trying to do this if u know anything about this process please post the code or link thank you

nagendra092918, 2019-09-11:
did any one implemented this process . please let me know and if possible provide what are the changes i need to know in code 
mail id: nagendra092918@gmail.com
thank you