thotro/arduino-dw1000 #203: crosstalk between devices

supersc, 2017-06-05:
Hi all:

I am working with 4 devices (2 anchors, 2 tags) to study the signal collision solution. And I seem to find some crosstalk issue. For example, I have anchor A talking only with tag 1, and anchor B talking only with tag 2. All transmission is with specific destination address. But somehow, any device can still receive the messages from the others... 

Has anyone seen this, or I am doing something wrong?

Thanks

StevenH2812, 2017-06-20:
I don't know the specifics of your implementation but obviously regardless of the destination address all antennas can receive the signal, since the physical EM wave travels outwards in all directions and not just specifically towards the antenna whose destination address you fill in. It is only after the message has been received (by all antennas) that the individual devices that receive something compare the destination address in the message to their own and decide whether the message was intended for them. In the library in this github the comparing of the addresses in the messages is done in the code (with the helper class DW1000mac) but I believe the DW1000 also has registry options that when enabled can do the MAC filtering automatically. If the pairs of anchors and tags remain constant then I think you can also avoid cross talk altogether by having your different pairs of anchors and tags operate on a different channel / frequency band.

r-schouten, 2017-12-12:
the rangingclass seems to be written for multiple anchors and tags, it have mac filtering implemented, but it don't work, in the dwm1000 class i saw that mac filtering was commented out, proberly is that the reason. but I'm not sure!

it should be possible, but to make it you have to make a few changes and you have to know how the code works. 

if you get it to work please upload your code, it will help some more people. I can't make it because I have not enough modules to test.