thotro/arduino-dw1000 #205: Questionable ranging performance: range dependent ranging error

StevenH2812, 2017-06-20:
I have been working on getting the ranging working with two DWM1000 modules. I am using arduino pro minis 3.3V 8MHz and have tried a variety of adapter boards (both adapter boards by localino, and adapter boards from Wayne's tinkering page). After some messing about (the one by Wayne's tinkering page does not work out of the box) all of the adapter boards are working properly with the DWM1000. 

However, when I get to the part where I actually analysis the ranging performance of the DWM1000 I am getting some peculiar results. The ranging error is dependent on the range itself. I know this is a known phenomena, hence why the receive timestamps are being corrected in the code through lookup tables, but for me the range error has much higher dependency on the ranging distance than those lookup tables account for.

I calibrate the ranging at 1 meter distance between anchor and tag by adjusting the antenna delay until the reported distance by the DWM1000 is almost exactly 1 meter. From this point I will increase the distance to 1.5 meters and down to 0.5 meters and 0.25 meters in order to measure the accuracy. As it turns out, the ranging error will increase and decrease depending on whether I increase or decrease the measurement distance with respect to the calibration point. So for example if it's calibrated at 1 meter it will report 1 meter at 1 meter measurement distance. Then if I measure at 1.5 meter it will report 1.6 meter, and when I measure at 0.5 meter it will report 0.4 meters.  This happens consistently with different adapter boards and DWM1000 modules. 

The effect seems to be worse when putting the DWM1000 into the 16 MHz PRF mode. In this mode the ranging error can be more than 30% of the deviation from the calibration point. In the 64 MHz PRF mode it will be about 10-20%. 

Here are some sample results. 
For the localino adapter, using 64 MHz PRF mode:

Actual range [cm] | Mean meas. range [cm] | Mean error [cm]  | Standard Deviation [cm] 
       25                  |           18.2                      |      -6.8                 | 2.030    
       50                  |           42.1                      |      -7.9                 | 1.901    
       100                |          101.4                     |      1.4                   | 2.063    
       150                |          161.2                     |      11.2                 | 1.454    

For the Wayne adapter, using 64 MHz PRF mode:
Actual range [cm] | Mean meas. range [cm] | Mean error [cm] |  Standard Deviation [cm] 
       25                  |           1.9                        |      -23.1              |  3.054    
       50                  |           29.4                      |      -20.6              |  2.645    
       100                |          94.9                       |      -5.1                |  2.250    
       150                |          152.6                     |      2.6                  |  1.545    

Note that the different adapter boards also have completely different DWM1000 modules, which is why the ranging results are different (I didn't recalibrate the antenna delay). However no matter what type of boards and modules I use the ranging bias appears to be about 10-20% of the deviation from the "calibration" point (or at least to the point where the measurement is giving no error). 

Some thoughts I had on potential causes:
- The frequency of the sampling clock used by the DWM1000 to determine timestamps is lower than the expected value, such that the timestamp resolution is not 15.65 picoseconds but actually higher. This would explain a range dependent error, but seems an unlikely cause given that I have tested multiple DWM1000 modules and all show similar results.
- The estimate of the received power of signals is incorrect, such that the lookup tables for range bias used in the code do not give the correct bias to correct with. The reason for me to mention this as a potential cause is that in the code an equation from the DW1000 user manual is used to estimate the receive signal power, whereas the lookup table is taken from the application notes APS011, which present a different equation to estimate the receive signal power which should only give signal power of the first unobstructed path, and not of the entire signal. I have not tested this yet, but it should not result in such a huge difference, since APS011 mentions that the range bias should only change about 3cm when the measurement range is increased by a meter, and I measure much higher errors.

I am at a loss as to what could be causing this. Anyone has any ideas as to what could possibly be causing this? Also, are others getting the same result, or are you not getting such a dependency on range?

kuek, 2017-06-21:
Hi,
nice you can use our Localino boards.

quick question, are the antennas (these ceramic antennas from the DWM1000) pointing upwards and not lying flat on a table?
When the signal propagation is disrupted heavily you will get these deviations.

regards,
kuek

StevenH2812, 2017-06-22:
Hi kuek, thank you for your reply. 

The antennas are not pointing upwards but "sideways" so to speak, because with the current setup pointing upwards would be very inconvenient (the cables etc. are attached at the bottom). So imagine both antennas pointing upwards and facing each other, but then rotate both antennas 90 degrees to one side and you will have my orientation during testing. So if you draw lines from the corners of one antenna normal to the plane of the antenna, then these lines will roughly intersect with the corners of the other antenna whose plane is parallel to the plane of the first antenna. Relative to each other it would therefore be exactly the same as having both pointing upwards, except that the surface they rest on is not below the antennas but to the side of them.

I make sure that they are not flat on the table because I already noticed that this affects the ranging measurements greatly. During all of my measurements they are always oriented the same way.

kuek, 2017-06-22:
how about longer ranges? just looked closer to your tables and measurement results. the performance does not look bad. even in close range like you did the measurements in. the mean error is always pretty low and in cm range. thats whats expected.

StevenH2812, 2017-06-23:
Hi kuek,

I was already planning on performing range measurements at longer ranges to see if the trend of range bias continues. The mean error is in the cm range, yes, but the troubling part is that it shows a clear dependency on the measurement distance. It is quite consistently showing a decreasing mean error when decreasing measurement distance, and vice versa. I feel like the ranging measurements should have zero mean error regardless of the ranging point, and a standard deviation in the centimeter range. But anyways, I will perform ranging measurements next week for longer ranges and see if the trend continues at larger ranges.

Also, the reason the Localino performs better in this case is because I calibrated the antenna delay on the localino, and didn't recalibrate the antenna delay for wayne's boards (they have 2 different decawaves with different antenna delay). If I recalibrate I'll probably get similar performance from wayne's boards haha.

VVHack, 2017-06-24:
I am a beginner in working with chips and electronic equipment so please pardon me for my equipment, I hooked up a  DWM1000 to an Uno, configured it to be an anchor, hooked up another one to a nano, configured it to be a tag, I get results like 1,5 m when they are 1 m apart, 2.4 m when they are 2 m apart.  Is this how it is supposed to be or am I doing something wrong?

StevenH2812, 2017-06-26:
@VVHack That sounds very similar to what I get as well, so you also get increasing range bias with increasing measurement distance. Not sure which of the ranging files you are using, but I found that the DW1000Ranging example gives slightly better ranging performance than the regular ranging example, but still shows increasing ranging bias with measurement distance (just less of an increase). This is likely caused by the fact that the DW1000Ranging example operates on the 64 MHz PRF mode. You can also enable this in the regular ranging example by changing the DW1000.enableMode line to DW1000.enableMode(DW1000.MODE_LONGDATA_RANGE_ACCURACY); (note: you will have to do that for both the anchor and the tag part of the example)

VVHack, 2017-06-28:
As an update, I tested it in better conditions, it was earlier lying on a reflective surface which might have messed with the signal, I kept the arduinos on a tripod and they gave much more accurate results

StevenH2812, 2017-07-07:
Update from myself. I tested the ranging at bigger measurement distances and initially still got the error scaling with measurement distance. However, after issue #143 I looked into the sign used to correct the the time stamp for range bias, and concluded that I also believe this sign is wrong. So I changed the timestamp correction in the correctTimestamp function from timestamp += adjustmentTime to timestamp -= adjustmentTime. This not only corresponds with what Decawave reported the sign should be for the range bias correction in APS011, but it also gives much better ranging results. Now I am able to get mean errors that stay within approximately +/- 15 cm for all distances tested. Just look at a barplot of the mean errors I am getting when using the postive correction vs the negative correction. The positive correction has clear scaling with measurement distance. The negative correction does not have a similar scaling and the error goes back down again at a certain distance.

[image]

closed 2017-08-01 by StevenH2812