thotro/arduino-dw1000 #206: Dwm1000 not working : Unique ID: FF:FF:FF:FF:FF:FF:FF:FF

bharath-lokanadhan, 2017-06-22:
### Step 1: What do you want?

* Report an issue.
* Improve code (and need help).
* Suggest a new feature.
* Present a new hardware adapter board or wiring.
* Want to report my experience or measurement. (Please also add them to the wiki!)

### Step 2: Describe your environment

 * Arduino IDE version or Toolchain: 1.83 mac
 * dw1000 Library version: latest as per today's date
 * Arduino device: arduino pro mini 3.3V 8Mhz
 * Used adapter board: no, bread board and powered by ftd1232
  
### Step 3: Describe the problem
 Actually i have 4 dwm1000 module except 1 dwm1000 module i am able to get the distance between two nodes

when connected this dwm1000  i am not getting anything  when using the examples from the library 
DW1000Ranging_ANCHOR  and DW1000Ranging_TAG.

so when used the basic connectivity test the results are below for the odd one dwm1000

Device ID: FFFF - model: 255, version: 15, revision: 15
Unique ID: FF:FF:FF:FF:FF:FF:FF:FF
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

the following is for rest of the three dwm1000
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 0A, Short Address: 05
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

Thank you,
BHARATH.

Rotzbua, 2017-06-24:
your wiring is wrong/incorrect. please search in the old issues, there are thousands of this type of reports. thanks.

closed 2017-06-24 by Rotzbua

michaelelandry, 2017-07-10:
Does anyone know why the "Unique" Identifier is always "FF:FF:FF:FF:00:00:00:00"?  It does not seem unique. because every chip I have gives the same result.  The user manual says it should be a globally unique identifier per the standard that is copied from OTP memory on every reset.   Assuming this is happening, why is decawave not putting a unique ID into OTP memory?

bharath-lokanadhan, 2017-07-11:
Check for wiring and dwm1000 pcb connections, and also for powering dwm1000.
Usually try powering separately powering dwm1000.
If still you get problems contact me.

bharathlamazing@gmail.com,
BHARATH LOKANADHAN,
INDIA.

onuralcaa, 2023-06-08:
How did you solve the problem?

onuralcaa, 2023-06-12:
I was using the stm32f1 processor and it worked when I changed the SPI modes from the library. Thank you.  Windows için Posta ile gönderildi Kimden: Nicolas SchurandoGönderilme: 12 Haziran 2023 Pazartesi 18:18Kime: thotro/arduino-dw1000Bilgi: onuralcaa; CommentKonu: Re: [thotro/arduino-dw1000] Dwm1000 not working : Unique ID: FF:FF:FF:FF:FF:FF:FF:FF (#206) How did you solve the problem?If I am not mistaken, the unique id register is intentionally left to the end user to program.However, all the values being 0xFF point towards a non working SPI communication with the chip. I would suggest as others did double-checking wiring and pinout.