thotro/arduino-dw1000 #217: 5V works but not 3.3V

VVHack, 2017-07-14:
I apologize if this has been asked before.  I am using an Arduino Nano with a DWM1000 to perform RX/TX communication and ranging.  Everything is working fine, I have observed repeatedly that when I hook up VCC to 5V, everything is working fine but when I use 3.3V, it mostly doesn't work.  Why is this the case even though the chip is rated for 3.3V?

Thank you

Rotzbua, 2017-07-15:
There is a lack of information to solve this mystery. I suggest you do not (or wrong) use a level shifter to convert 3.3V logic level of DWM to 5V Arduiono.

VVHack, 2017-07-17:
I have not been using any logic level converter, just directly hooking up the Arduino to the chip, what do you think could be the issue?

Rotzbua, 2017-07-17:
3.3V can not communicate with 5V without a level shifter. The missing level shifter is the issue. Add an level shifter and it is solved.

closed 2017-07-17 by Rotzbua