thotro/arduino-dw1000 #218: Have someone successfully connect 1 tag with 4 anchors?

jack880019, 2017-07-16:
Hello! I am using this lib for indoor positioning project and now I can connect 1 tag with 3 anchors. And now I want to connect with 4 but it seems not working. Have someone successfully connect 1 tag with 4 anchors? I using the example code DW1000Ranging_ANCHOR. Plz give me some advise thx

fatihpc, 2017-08-14:
Yeah. I wonder if this is the answer. Do you work with 4 anchors? Or more ...

werty37, 2017-09-15:
I have a very basic question. Can you read the distance of a tag from all the anchors through only a single anchor?

prideofisland, 2017-09-24:
> Have someone successfully connect 1 tag with 4 anchors? 

Hi @jack880019 
Of course, it is possible. I've tried it once, but never used it afterwards.
Check this line:
In DW1000Ranging.cpp
`//Max devices we put in the networkDevices array ! Each DW1000Device is 74 Bytes in SRAM memory for now.
#define MAX_DEVICES 4`

You have to increase this number. 

Piotr

Rotzbua, 2017-09-24:
@pascalbros Thanks. I close this.

closed 2017-09-24 by Rotzbua