thotro/arduino-dw1000 #220: Connecting DWM 10000 with esp 8266 Lo-Lin

hdineth, 2017-07-21:
Hello,
I am working with an indoor localization project with DWM 1000 module. I need to know the method of connecting DWM 1000 with ESP 8266 module including
1. Pin connections
2. Code. (I use Arduino software to programme ESP 82666)

I tried to understand some community posts. But i cannot get it properly.
Thanks.

Rotzbua, 2017-07-21:
Please use google or youtube to find tutorials to understand the ESP and Arduino IDE. Also you have to understand SPI and (Hardware)interrupts. Best.

closed 2017-07-21 by Rotzbua

hdineth, 2017-07-21:
sir, i do know basics on esp8266 and arduino ide.i need to know on how to
connect esp8266 and dwm1000 via spi.

curlyz, 2017-07-23:
Buy a dev kit from kuek, Localina. Nobody gonna help you with the ESP, try to understand that you must port it yourself to another MCU that is not supported by this project.

werty37, 2017-12-30:
@hdineth I have been able to connect DWM1000 to Wemos D1 Mini. These are the pin mappings:

DWM1000 Adapter - ESP8266 WeMos D1

RST - D2
IRQ - D1
CLK - D5
MISO - D6
MOSI - D7
CS - D8

Hope this helps.

ricgaldame, 2021-12-13:
Hello i am having the same issue with esp8266. Please somebody have a step by step about it?