thotro/arduino-dw1000 #221: Doesn't work with Teensy 3.2 at greater than 48Mhz CPU Speed.

GremlinsInGame, 2017-07-22:
* Arduino IDE version or Toolchain: 1.6.9,   teensyduino 1.35
 * dw1000 Library version: master.. ?
 * Arduino device: Teensy 3.2
 * Used adapter board: No
  
None of the Examples seem to work with Teensy 3.2 if using any CPU Speed greater than 48MHZ.
Default speed on Teensy 3.2 is 72MHz which causes things not to work.

BasicConnectivityTest is deceiving because it seems to report back info from the DW1000 even if SPI is not connected??  This is a seperate issue but causes troubleshooting to be very difficult.

Everything works on Teensy LC at its standard speed of 48 MHz.

Is there code in the Library that causes problems based on Host CPU speed?
I can't seem to find a work around.

Rotzbua, 2017-07-23:
SPI (and GPIOs) is highly hardware dependent. The arduino stuff like `#include <SPI.h>` are just a abstraction to simplify the hardware usage. So in conclusion: yes, different CPU speed can break a working library depending on teensy implementation of `#include <SPI.h>`

cimbalo, 2017-07-31:
Have you tried to set a SPI rate [here](https://github.com/thotro/arduino-dw1000/blob/master/src/DW1000.cpp#L105) that is a divider of 72MHz? 18Mhz should be ok.
Remember DW1000 doesn't support SPI rate above 20MHz.

closed 2017-08-12 by Rotzbua