thotro/arduino-dw1000 #227: Using 3 Anchors and 1 Tag, to triangulate position

TempleClause, 2017-08-18:
I would like to setup 3 Anchors let's call them A, B, C and one Tag let's call it T.
Tag A would be connected to a PC and visualize the position of Tag T. 
I see two approaches to do this:

Approach 1:
Send messages from Anchors B and C to Anchor A containing the distance to Tag T. 

Approach 2: 
Send a Message from Tag T to Anchor A containing the distances to Anchor B and C. 

What would be the preferred approach and why? 

How should I address this problem? I got the Anchor and Tag as well as the Messaging Ping Pong examples to work but they do not seem to be combinable. My approach would be to modify the DW1000Ranging.cpp file to extend it so it can send regular messages as well. 

How would you guys do this? Any hint in the right direction is much appriciated!

Thanks

ZizziWang, 2017-09-10:
Hello guys,

I am also working on it but with 8 anchors and 1 tag.
But I am just a noob, so please let me know if there's any mistake.
I prefer the approach 2:
  Tag T sends a string including all distances to all anchors as a broadcast message.
Not only anchor A but also B & C can visualize the position of Tag T.

My reason is approach 1 will make anchor B & C become senders, it will
  1. make the firmware more complex
  2. take more time to reset the status

However I am now facing a BIG problem:
  transmit() & receive() doesn't work after loop(), I received a garbled message...
I am now trying to add a function in the DW1000Ranging.cpp to send the broadcast message after RANGE_REPORT.
Does anyone have better idea about this issue?
Any hint will be appreciated.

Many thanks

prideofisland, 2017-09-23:
> What would be the preferred approach and why?

Hi @TempleClause 

How about introducing a new unit that will be obligated to collect the measurements and transmit it to the PC? I've used this approach in my project and I didn't regret this decision. I called this unit a 'control unit' (CU in the picture below). 

pros:
- you don't have to change code for anchors and tag
- CU only receives the results (do not 'participate' in distance measurement). It means that you can randomly move your 3 anchors (one anchor don't need to be connected with your PC). 

cons:
-you need a one more DW1000 chip

[image]

kerr4huo, 2017-09-29:
@ZizziWang 
To achieve ranging with larger number of anchors, a few modifications might need to be taken care of:
a. MAX_DEVICES
b. deltaTime uint16_t -> uint32_t
c. possibly, useExtendedFrameLength.

Although I achieved ranging from 1 tag to 8 anchors, I did heavy modifications for polling a distance matrix among all the UWB modules. Thus, my code is super messy now. I will try to clean it up and create a branch when I become less busy. Hope this comment helps. @ZizziWang

kerr4huo, 2017-09-29:
Also, I think it's better to discuss the multiple anchors issue separately from this thread. Should we create new discussion thread? @ZizziWang

ZizziWang, 2017-10-01:
@kerr4huo 
Your comment did me a great help.
I tired 1 tag + 4 anchors before and result in 3 available distances and 1 negative distance.
I guess it is caused by the LEN_DATA before.
But now I find it is more complex than I though...
It's better to open a new discussion thread for this issue.
Can you do that and share the details you found?
Thanks again.

Archeresque, 2017-10-26:
@prideofisland I'm having trouble with the code for a receiver to collect the data from the tag. I tried implementing the send and receive example into the code of the tag but my output from the receiver is garbled. Any ideas?

hdineth, 2017-11-06:
try using external power supply with at least 1A output for anchors. Do not connect to USB

az-z, 2017-11-06:
1A? Like one amper? Sounds a lot..

hdineth, 2017-11-07:
yes sounds a lot.but normally when a transceiver device in action,it
consume s more current.also arduino, wires drain some current too

anthony211212, 2024-05-11:
How can i implement this library for multiple anchors and tags? I could not find deltaTime and frame length option. I also checked all issues in this repository. Can you explain in more detail?

> @ZizziWang To achieve ranging with larger number of anchors, a few modifications might need to be taken care of: a. MAX_DEVICES b. deltaTime uint16_t -> uint32_t c. possibly, [...]

anthony211212, 2024-05-11:
> @ZizziWang To achieve ranging with larger number of anchors, a few modifications might need to be taken care of: a. MAX_DEVICES b. deltaTime uint16_t -> uint32_t c. possibly, [...]

How can i implement this library for multiple anchors and tags? I could not find deltaTime and frame length option. I also checked all issues in this repository. Can you explain in more detail?