thotro/arduino-dw1000 #228: Dw1000 wrong distance and calibration

fatihpc, 2017-08-19:
How do I calibrate the dw1000 module? Will we do calibration for anchor and tag? Or are we going to make arrangements in the library? I measure 274 cm with my anchorum. But the actual distance is 196.5.

kerr4huo, 2017-08-19:
Same issue here. For now I just subtract a constant to compensate. I am wondering and searching what's the cause of this problem and how to fix also. Thanks for bringing up this question. Will update if I have anything new.

fatihpc, 2017-08-20:
I think this is a chip fiasco. Because every time I switch the device on and off, the metrics change. So distance travel is not static but dynamic. That's why I can not develop a formula. I expect help from the master.

werty37, 2017-09-11:
I have the same issue. It error varies between 0.5 to 0.75 meters. Do you have any idea how i can correct it please?

prideofisland, 2017-09-23:
> Do you have any idea how i can correct it please?
Hi @werty37 
In my project I used a practical approach. I've measured the error for various distances (2-20m). Each measurement is represented by red plus symbol. 

> The error varies between 0.5 to 0.75 meters.
You're right here. I have the same issue. In general, I would say that DW1000 chip is inaccurate but quite precise (the results are repeatable).  So it's not that bad ;) 

Using this points I generate a function. I called it 'correction function'.  I used 'polyfit' function in MATLAB to get this blue line on the graph.

Now we can finally make an use of the correction function. 
1. DW1000 measures 7,65m 
2. The measured value is the input of correction function. Correction function calculates it. The output is about 0,79m. 
3. The DW1000 chip substracts 7,65-0,79 and displays 6,86m. 
[image]

>  I just subtract a constant to compensate. 
@kerr4huo 
My method is an extension of your idea. It does NOT remove the measuring error but help to reduce it.

Piotr

werty37, 2017-09-26:
Hi @prideofisland 

That is a good idea. I will try to follow proper anchor positioning and then try your method. As of now i am trying to make the Ranging Anchor code on an ESP8266 since i will do the localization on the PC. If i am able to port it successfully, i will create a wiki/howto. Lets see. 

Thanks
S

hdineth, 2017-09-28:
hello sir,
i have created 3 anchor modules and a tag and everything works fine. Whn i connect the anchors and tag, my tag recieves a flow of data like in the attached picture from each anchor at a time sequentially. All i want to do is to store these values in a 3by 1 matrix like as follows,
`float mat[3][1];
mat[0][0]=value from 1st anchor node;
mat[1][0]= value from 2nd node;
mat[2][0]= value from 3rd node;

`
likewise. but when i initilaize and save values to the matrix (inside the newRange function) it saves like follows,
mat[0][0]=value from 1st anchor node;
mat[1][0]= value from 1st node;
mat[2][0]= value from 1st node;

are there any way to correct this issue.?
[image]