thotro/arduino-dw1000 #230: Use with ESP32

gabdsg, 2017-09-18:
I'm trying to use with an ESP32 and get this error:
```
DW1000.cpp:172:6: error: 'class SPIClass' has no member named 'usingInterrupt'

  SPI.usingInterrupt(digitalPinToInterrupt(irq)); // not every board support this, e.g. ESP8266

exit status 1
Error compilando para la tarjeta ESP32 Dev Module.
```

Need help, thanks

Rotzbua, 2017-09-19:
@gabdsg Sorry, but if you want to program embedded systems you have to know the basics about c++ and the arduino hardware. This is not a 'let a led blink' library ;)
For your c++ problem:
https://lmgtfy.com/?q=c%2B%2B+class+has+no+member+named+using

Otherwise you have to look in to the source code in the arduino repository and compare them with the source code of the esp32 library.

jiaolongbing, 2020-03-23:
> I can't find anything about the esp32 and dw1000

The the same to you.

jiaolongbing, 2020-03-23:
> @gabdsg Sorry, but if you want to program embedded systems you have to know the basics about c++ and the arduino hardware. This is not a 'let a led blink' library ;)
> For your c++ problem: [...]

I want to program embedded systems by python, are there something for the idea!?

wzqvip, 2023-05-07:
Which ESP32 support this "error: 'class SPIClass' has no member named 'usingInterrupt'"
I tried on ESP32-PICO-D4, ESP32-C3. They both doesn't have it.
What will it cause if I just comment this command?