thotro/arduino-dw1000 #235: Arduino Nano_decawave ranging issue

SIVAComtustec, 2017-10-12:
hi, im working on arduino nano with decawave for ranging application, im using the code from github. i have tested all th examples in the file, but im not getting any ranging result. can any one please let me know where im struggling, herewith i attached the pin connections and code and result . pls review and giv ur input.(https://github.com/thotro/arduino-dw1000/)...

my pin connection:
d2- irq
d9-rstn
miso-11
mosi-12
cs-10
clk-13
power supply for decawave from external dc supply.

my serial terminal output:

### DW1000-arduino-ranging-tag ###
DW1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: EUI: 0:0:0:0:255, OUI: 255:255:255
Network ID & Device Address: PAN: 10, Short Address: 2
Device mode: Data rate: 110 kb/s, PRF: 16 MHz, Preamble: 2048 symbols (code #4), Channel: #5

the output struck here and no result anymore after this

Rotzbua, 2017-10-12:
` Device ID: DECA - model: 1, version: 3, revision: 0` seems that your connection from arduino to dw1k is good. Must be something else.

SIVAComtustec, 2017-10-13:
hi , i have to ranging output after i gave separate power to decawave with small wire. but i have error reading,,distance changine with error. even when i kept a tag closer to anchor it showing 0.40 m distance, whether i should hav antenna ,,pls support me,,any algorithm shold use in code, alraeady assymtric ranging algorithm is used in the code,but stil error coming

closed 2018-02-14 by Rotzbua