thotro/arduino-dw1000 #237: LED Blink Feature

werty37, 2017-10-20:
### Step 1: What do you want?

* Improve code (and need help).

### Step 2: Describe your environment

 * Arduino IDE version: 1.8.3
 * dw1000 Library version: latest
 * Arduino device: Pro Mini 3.3v
 * Used adapter board: Wayne Breakout Board
  
### Step 3: Describe the problem
How do you enable RX and TX Led blinking during ranging?
  
#### Observed Results:

  * What happened?  This could be a description, log output, etc.
I looked at the DW1000 class and added the following code in setup. But i am missing something i guess.   

#### Expected Results:
Blink TX and RX LED during Ranging.
 
#### Relevant Code:

```arduino
  // TODO(you): code here to reproduce the problem
```
void setup() {
// ... 
  DW1000.enableDebounceClock();
  DW1000.setGPIOMode(MSGP2, MSGP3);
  DW1000.enableLedBlinking();
//...
}

Rotzbua, 2017-10-20:
Wrong order? See #144

werty37, 2017-10-20:
Thanks. The code seems to compile without any errors but i am somehow not able to get the intended behavior. I am using Wayne breakout board; Pins 12 and 13  (GPIO 3 / TXLED, GPIO 2 / RXLED) from the DWM1000 are soldered properly to the Wayne board with 0603 smd led and 1k resistors.

Rhodes187, 2017-12-12:
Same here, I can successfully implemetrent BasicSender and verify packet sending, but the TXLED wont turn on

Rotzbua, 2017-12-13:
Then fix it and make a pull request...

werty37, 2017-12-29:
Okay, I have been able to get it to work. But somehow it seems to be at random even though it is not doing any ranging. 

```cpp
DW1000.enableDebounceClock();
DW1000.enableLedBlinking();
// enable GPIO2/RXLED blinking
DW1000.setGPIOMode(MSGP2, LED_MODE);
// enable GPIO3/TXLED blinking
DW1000.setGPIOMode(MSGP3, LED_MODE);
```
Anyone else has the same issue?

Rotzbua, 2017-12-29:
The dw1000 is a network device. RX and TX shows input and output, which do not depend on any "ranging".

werty37, 2017-12-30:
Thanks for clarifying @Rotzbua . Adding to that there are several configurations after looking at dw1000_user_manual_v2.05 PG 122, which can be done

`DW1000.setGPIOMode(MSGP2, LED_MODE);`

MSGP0: 
When operating as the RXOKLED driver, the output is asserted briefly when the receiver 
completes the reception of a frame with a good FCS/CRC. 

MSGP1:
When operating as the driver, the output is asserted briefly when the receiver detects the 
SFD sequence  in  the  RX  frame.

MSGP2:
When operating as the RXLED driver, the output is asserted when the receiver is on, and stays 
on for a brief period after the receiver is turned off.  

MSGP3:
When  operating  as  the TXLED driver, the output is  asserted  briefly  when  the  transmitter 
completes  sending  a  frame.

MSGP4, MSGP5, MSGP6:
The pin operates as the EXTPA / EXTTXE / XTRXE output.

MSGP7, MSGP8:
The pin operates as GPIO7 and GPIO8 respectively when LED_MODE is set. 

Hope this helps other.

Rotzbua, 2017-12-30:
Thanks for research 👍 , please can you copy your collected and summarized information to wiki? The wiki can be easily edited by button top right "edit". The wiki is open for everybody to write and edit.

werty37, 2017-12-30:
@Rotzbua, Updated the wiki. Please check and add/correct if necessary. 
https://github.com/thotro/arduino-dw1000/wiki/DW(M)1000-LED-Blink-Feature

Rotzbua, 2017-12-30:
@werty37 Thanks a lot!

closed 2017-12-30 by Rotzbua