thotro/arduino-dw1000 #238: Issue with DWM1000 connecting multiple anchors with one tag

Archeresque, 2017-10-21:
[DW1000Ranging_ANCHOR_final.zip](https://github.com/thotro/arduino-dw1000/files/1404492/DW1000Ranging_ANCHOR_final.zip)
[DW1000Ranging_TAGfinal.zip](https://github.com/thotro/arduino-dw1000/files/1404493/DW1000Ranging_TAGfinal.zip)

Hello, my goal is to track a tag using trilateration with 3 anchors. So far I have been unsuccessful getting multiple DWM1000 Modules to communicate. I am using the latest release of this library and the 1.81.10.0 release of the Arduino IDE. I assembled my units as per Mr. Wayne Holder's guide located [here.](https://sites.google.com/site/wayneholder/uwb-ranging-with-the-decawave-dwm1000---part-ii) I built my units and I have them transmitting in a pair without issue after adjusting the library to fit Mr. Holder's design. Now I am trying to get everything to communicate as described by @reycreator17 in this [thread.](https://github.com/thotro/arduino-dw1000/issues/179) but even using his library and code I am unable to get multiple to communicate. Here is what I see when I try to run multiple:
[image]
When I run the basic connection test everything appears to be fine and the regular ranging test works. 
[image]
[image]

Does anybody have any ideas about what I should change in the library or code to get multiple anchors and one tag to work? I've tried many different troubleshooting methods but I seem to be at an impasse. 
Lastly, I've attached the code I am using in case that is the issue there. Thank you for any help or recommendations!

closed 2017-10-22 by Archeresque

Ritesh23, 2018-03-13:
Hi @Archeresque , 

Have you managed to solve using 3 anchors with 1 tag?

asheeshraikwar, 2019-05-11:
please ..anyone tell me the steps to modified the library for using multiple anchors and one tag ?

reyfrancis, 2019-05-15:
Hi. I'm using @reycreator17 before. I changed email recently. 

I haven't touched my DWM1000 for a long time. 

@Archeresque
@Ritesh23
Why don't you try using the example I uploaded and the old Arduino version

@asheeshraikwar 
What do you mean by modified library?