thotro/arduino-dw1000 #239: Building a Receiver for a System of 3 Anchors and 1 Tag

Archeresque, 2017-10-26:
I have a system of 3 anchors that are tracking one tag. The problem with this system is that the tag is wired to my computer and I'd like it to be wireless. I decided to build a receiver that would then communicate with the tag and push the distance data from the tag through the serial port of my computer. I tried implementing the send and receive examples into the code of my tag and receiver but I am receiving a garbled output. My system uses Arduino Pro Mini and the latest version of the Arduino IDE as well as the latest release of this library. Any ideas on how I should move forward to program my receiver?

r-schouten, 2017-10-28:
I will try the same thing next week. I have multiple anchors wich have to send the data to one master anchor wich is connected to my computer or wifi. when I succeed I will shared my code with you.

please keep me updated

Archeresque, 2017-10-28:
@r-schouten I haven't completely solved the problem but I was able to mitigate a temporary solution. Since I was able to send the data packet out amongst a garbled mess, I simply programmed the receiver to ignore and random characters and only read the data I'm looking for. My receiver output looks like this:
1,1.19
liufliwuefhuefhpu
cwieufhvoiweufhvi
viweufhiweufhviuef

So I just had it look for the anchor ID and parse from there.

r-schouten, 2017-10-31:
can you upload that code?

Archeresque, 2017-10-31:
@r-schouten 
Here is a screenshot of the function that will read the packet. It's very basic. If you have your sending unit format its output as a string, just use an ID number, a delimiter, and then the data and you'll have an easy to read packet. 
[image]

hdineth, 2017-11-06:
[image]
[image]

[TAG.zip](https://github.com/thotro/arduino-dw1000/files/1446122/TAG.zip)

Use above code in tag. Before use it, you should change anchor values to static (eg. 1 2 and 3). In order to do so , you have to edit dw1000device.cpp and dw1000ranging.cpp files. refer the attached images to edit them.

Archeresque, 2017-11-09:
@hdineth Very clean tag code and good explanation of the anchor addressing.

Rhodes187, 2018-02-12:
@Archeresque I will be doing the same implementation, but I'm still kind of stuck at getting three anchors to work at the same time. Did you use something like the token ring protocol?

anthony211212, 2024-05-11:
> [image] [image]
>
> [TAG.zip](https://github.com/thotro/arduino-dw1000/files/1446122/TAG.zip)
>
> Use above code in tag. Before use it, you should change anchor values to static (eg. 1 2 and 3). In order [...]

How can i implement this library for multiple anchors and tags? I could not find deltaTime and frame length option. I also checked all issues in this repository. Can you explain in more detail?