thotro/arduino-dw1000 #240: Setting up delay time in Arduino TAG code

hdineth, 2017-10-31:
I am using 3 anchores with one tag. Tags has given hardcoded as 1,2, & 3. I receive measurements from 1,2 &3 in order. Also I have impelemted a least square algorithm for better accuracy. I attached the algorithmin to the TAG code. The measurement process is as follows.

1. TAG receives 3 distance measurements in order . ( from anchor 1,2 & 3)
2. Then these measurements saved into a 3x1 matrix.
3. Then  above matrix is used for the calculations of the least square algorithm.
4. Then algorithm calculates the actual position.

Th issue i have is when algorithm is running in the TAG code, It does not run smoothly. It does not provide position continuously.  I think the issue is with the delaying process. When i add a delay time in TAG code (e.g, delay (1000)) it starts to misbehave and does not provide measurements continuously. 
I think when adding the algorithm, processor (pro mini ATMEGA 328) cannot handle the calculations with the receiving speed of the data . Are there any way to slow down the measurements receiving speed from the anchors? ( to at least one measurement set for 30 seconds)

hdineth, 2017-11-07:
@thotro @frwck can you please help

werty37, 2017-11-07:
@hdineth I had the same issue even with TWR using Arduino Pro Mini. So i moved to ESP with the anchors running at 80mhz and the tag running at 160mhz. The localization is done on a PC though.

hdineth, 2017-11-08:
@werty37  How do you change the mode to TWR? Also what is mean by 80mHz and 160 mhz? How did you change it? I tried to setup delay time by changing DEFAULT_DELAY_TIME in dw1000ranging.h. Did you try it?

werty37, 2017-11-10:
@hdineth I haven't check the update frequency of TWR itself but i changed the clock speed of ESP. By default it runs at 80Mhz but you have an option of changing it to 160Mhz. I am using Arduino IDE to do the same. Check "Tools > CPU Frequency" in Arduino IDE.

werty37, 2018-01-15:
@hdineth I think what you are looking for is dw1000.deepSleep() method.

darshb34, 2018-06-13:
@hdineth  how did you implement least square for this positioning application? I have used trilateration algorithm and I am getting MSE of +-20cms.