thotro/arduino-dw1000 #245: extending the range of the dwm1000

r-schouten, 2017-11-28:
The dwm1000 modules only works within a range of 10 meters. That is very short because the dwm1000 product brief and the dw1000 datasheet says that it will work far over 100 meter outside and 20-50 meters inside a building.

i've tried to change the channel and the mode of the dwm1000, but it doesn't make the range longer.

does anyone know a solution?

kuek, 2017-12-03:
hi,
it depends on many many factors. its hard to answer this remotely. as I don't know your hardware, neither your firmware, nor your chip settings.
a general answer: if the RF signal is not well you will have low range only... and this may also depend on hardware around the chip.

kuek

r-schouten, 2017-12-06:
hi,

i've found a solution that make the range a lot bigger, we have reached a range of 200m on open terrain and 30 meters inside a building.

the signal transmit strength is configurable in the dw1000 chip according to the dw1000 user manual "7.2.31 Register file: 0x1E –Transmit Power Control". in the dw1000.cpp file this transmit strength caliblation is calibrated for every channel. this calibration is not right. i've set the transmit strength so high as possible. I don't understand why we need this calibration.

so I've replaced 
`if(_channel == CHANNEL_1 || _channel == CHANNEL_2) {
		if(_pulseFrequency == TX_PULSE_FREQ_16MHZ) {
			if(_smartPower) {
				writeValueToBytes(txpower, 0x15355575L, LEN_TX_POWER);
			} else {
				writeValueToBytes(txpower, 0x75757575L, LEN_TX_POWER);
			}
		} else if(_pulseFrequency == TX_PULSE_FREQ_64MHZ) {
			if(_smartPower) {
				writeValueToBytes(txpower, 0x07274767L, LEN_TX_POWER);
			} else {
				writeValueToBytes(txpower, 0x67676767L, LEN_TX_POWER);
			}
		} else {
			// TODO proper error/warning handling
		}
	} else if(_channel == CHANNEL_3) {
		if(_pulseFrequency == TX_PULSE_FREQ_16MHZ) {
			if(_smartPower) {
				writeValueToBytes(txpower, 0x0F2F4F6FL, LEN_TX_POWER);
			} else {
				writeValueToBytes(txpower, 0x6F6F6F6FL, LEN_TX_POWER);
			}
		} else if(_pulseFrequency == TX_PULSE_FREQ_64MHZ) {
			if(_smartPower) {
				writeValueToBytes(txpower, 0x2B4B6B8BL, LEN_TX_POWER);
			} else {
				writeValueToBytes(txpower, 0x8B8B8B8BL, LEN_TX_POWER);
			}
		} else {
			// TODO proper error/warning handling
		}
	} else if(_channel == CHANNEL_4) {
		if(_pulseFrequency == TX_PULSE_FREQ_16MHZ) {
			if(_smartPower) {
				writeValueToBytes(txpower, 0x1F1F3F5FL, LEN_TX_POWER);
			} else {
				writeValueToBytes(txpower, 0x5F5F5F5FL, LEN_TX_POWER);
			}
		} else if(_pulseFrequency == TX_PULSE_FREQ_64MHZ) {
			if(_smartPower) {
				writeValueToBytes(txpower, 0x3A5A7A9AL, LEN_TX_POWER);
			} else {
				writeValueToBytes(txpower, 0x9A9A9A9AL, LEN_TX_POWER);
			}
		} else {
			// TODO proper error/warning handling
		}
	} else if(_channel == CHANNEL_5) {
		if(_pulseFrequency == TX_PULSE_FREQ_16MHZ) {
			if(_smartPower) {
				writeValueToBytes(txpower, 0x0E082848L, LEN_TX_POWER);
			} else {
				writeValueToBytes(txpower, 0x48484848L, LEN_TX_POWER);
			}
		} else if(_pulseFrequency == TX_PULSE_FREQ_64MHZ) {
			if(_smartPower) {
				writeValueToBytes(txpower, 0x25456585L, LEN_TX_POWER);
			} else {
				writeValueToBytes(txpower, 0x85858585L, LEN_TX_POWER);
			}
		} else {
			// TODO proper error/warning handling
		}
	} else if(_channel == CHANNEL_7) {
		if(_pulseFrequency == TX_PULSE_FREQ_16MHZ) {
			if(_smartPower) {
				writeValueToBytes(txpower, 0x32527292L, LEN_TX_POWER);
			} else {
				writeValueToBytes(txpower, 0x92929292L, LEN_TX_POWER);
			}
		} else if(_pulseFrequency == TX_PULSE_FREQ_64MHZ) {
			if(_smartPower) {
				writeValueToBytes(txpower, 0x5171B1D1L, LEN_TX_POWER);
			} else {
				writeValueToBytes(txpower, 0xD1D1D1D1L, LEN_TX_POWER);
			}
		} else {
			// TODO proper error/warning handling
		}
	} else {
		// TODO proper error/warning handling
	}`
in this simple line of code
`    writeValueToBytes(txpower, 0x1F1F1F1FL, LEN_TX_POWER);
`

I hope that this helps for everyone that need a longer range! and I want to know why this calibration is made.

az-z, 2017-12-06:
Looks awesome. Maybe a patch?

kuek, 2017-12-06:
well, "patch". hm. make sure you stay in the regulatory limits by your country, which are usually -41.3 dBm/MHz PSD. otherwise regulatory may pay you a visit... and this is usually not cheap.
instead of increasing output power, make sure your recevier is more sensitive, your have high receiver gain, hardware is right. power is not always a solution, instead it will mostly cause harm and not fulfill regulations.

az-z, 2017-12-06:
Good point. My oversight. Sorry.

r-schouten, 2017-12-07:
i measured a signal strength of -67db, that is far away of -41.3db

r-schouten, 2017-12-07:
the calibration used in the library is copied from table 18 and 19 of the dw1000 user guide. the dw1000 is the chip inside the dwm1000 module, I think the dwm1000 module have an other antenna than decawave have used to make this calibration.

if the maximum tx power is too high we should make an new calibration.

kuek, 2017-12-07:
its -41.3 dBm/MHz. this is a power spectral density. 
**-67 dBm TX EIRP. you stated 200m that you reached with that settings. lets consider at 3.5 GHz thats 93 dB loss, so your hardware has a sensitivity of -160dB. Clearly impossible!**

-67 dBm (which you probably mean) with your power settings, then either you have measured wrong or your hardware is odd. 

just a warning, if regulations catch you with this, make sure you have a good explanation.

r-schouten, 2017-12-07:
is there a way we can make an new calibration, is there any way we can measure the real signal strength? this calibration was copied from the dw1000 user guide and not made for this specific antenna. 

in the dwm1000 product brief i saw this "_Excellent communications range of up to 300m thanks to coherent receiver techniques*_", why would they say that if only a range of 10m is permitted?

r-schouten, 2017-12-12:
i've think i've found the answer in the dwm1000 datasheet.

To achieve best results when using the DWM1000 with Deca w ave’s DecaRanging software , you will need to adjust the default transmit power value programmed into the DWM1000 by the software. This is because DecaRanging software is targeted at Decawave ’s EVB1000 evaluation board which has a different RF path compared to the DWM1000. You should increase the transmit power by approximately 3 dB.

so, if we add 3dB to the calibration we use that should be oke right?

mao93cy, 2018-03-05:
Hi, I want to set the PSD to -41.3dBm / MHz according to the regulation of my country. However, I am not sure how and where to set it. Can someone explain the meaning of the hex values in "writeValueToBytes"?

skadiexe, 2018-03-06:
dw1000 user manual: 
[image]

nxdefiant, 2018-07-31:
With the default settings I get a maximum range of about ~4m for the dwm1000 modules, so I did a short test with gain=33.5dB.resulting resulting in a maximum range of ~10m outdoor. Looks like the advertised maximum range is a bit too exaggerated..

an4cr0n, 2018-07-31:
Me and @Sonic0 got the same results with the default settings, try to activate smart power and see if it gets better. We'll make an update as soon as possible.

nxdefiant, 2018-07-31:
Already tested smartpower, gave me ~2m. 

Btw can someone please explain me why "MODE_LONGDATA_FAST_LOWPOWER" sends with more power then "MODE_LONGDATA_RANGE_ACCURACY"?

MODE_LONGDATA_FAST_LOWPOWER sets PRF to 16MHz, on channel 5 the txpower is set to 16dB. MODE_LONGDATA_RANGE_ACCURACY sets PRF to 64MHz, on channel 5 the txpower is set to 8.5dB.

an4cr0n, 2018-07-31:
Check the 2nd comment to this issue. I think that's the thing you are
looking for.

Sonic0, 2018-08-17:
@nxdefiant remember that SmartPower is useful when you use short messages(short preamble). See Application Note APS023 from Decawave.

nxdefiant, 2018-08-19:
Strange with same settings I got 20m or more today. I haven't changed anything in the setup. The only thing that notably changed is the weather..

I'm testing with the Localino v1.3 modules btw.

SATHYALAKSHMI123456, 2019-06-22:
can any one tell how to increase the dbm?

Sonic0, 2019-06-23:
@SATHYALAKSHMI123456
In this driver is not disponible a method to set manually the transmission power.
You should write the registry directly or modify functions as in the beginning of this issue.

SATHYALAKSHMI123456, 2019-06-24:
can you tell me a some syntex

Sonic0, 2019-06-24:
> can you tell me a some syntex

I recommend you to use a fork of this library, Arduino-sw1000-ng (search on GitHub but it is very different from this). There are two different public functions with witch you can set up the transmission power.
You should read the user manual (a transmission power part) before using this setting.

SATHYALAKSHMI123456, 2019-06-24:
IF U HAVE A CODING FOR THIS PLEASE SEND ME .WE ARE OBTAIN 50MTRRS

Sonic0, 2019-06-24:
Please do not write in Uppercase. 
As in the third comment of this issue, I think you can call _writeValueToBytes(txpower, <powerInHex>, LEN_TX_POWER);_ . See user manual to insert the desired value.

SATHYALAKSHMI123456, 2019-06-25:
ok sir.i cant understand please help me