thotro/arduino-dw1000 #246: DWM1000: Network ID and Device Address Incorrect

Ritesh23, 2017-11-28:
I am using the BasicConnectivityTest example from the latest thotro arduino-dw1000 library and the Decawave DWM1000 chip to test whether the device ID, network ID and unique Identifier are being set as intended.

I am using Arduino IDE v1.8.3 along with the Arduino Uno connected as follows (the Uno pins I am using are the same as the ones used in the image below for the Arduino Pro Mini):
[image]

The issue I am having is that the PAN is displaying as 00 and the Short Address as FF. I have not changed anything from the code (apart from adding a Serial.println later towards the end of the code). The output when running the code is as follows:
[image]
  
What I am expecting as an output is that the PAN displays as 0A and the Short Address as 05, as is set in the BasicConnectivityTest example code shown below:
  
-----------------------------------------------------------
#include <SPI.h>
#include <DW1000.h>
// connection pins
const uint8_t PIN_RST = 9; // reset pin 
const uint8_t PIN_IRQ = 2; // irq pin
const uint8_t PIN_SS = SS; // spi select pin
void setup() {
  // DEBUG monitoring
  Serial.begin(9600);
  // initialize the driver
  DW1000.begin(PIN_IRQ, PIN_RST);
  DW1000.select(PIN_SS);
  Serial.println(F("DW1000 initialized ..."));
  // general configuration
  DW1000.newConfiguration();
  DW1000.setDeviceAddress(5);
  DW1000.setNetworkId(10);
  DW1000.commitConfiguration();
  Serial.println(F("Committed configuration ..."));
  // wait a bit
  delay(1000);
}
void loop() {
  // DEBUG chip info and registers pretty printed
  char msg[128];
  DW1000.getPrintableDeviceIdentifier(msg);
  Serial.print("Device ID: "); Serial.println(msg);
  DW1000.getPrintableExtendedUniqueIdentifier(msg);
  Serial.print("Unique ID: "); Serial.println(msg);
  DW1000.getPrintableNetworkIdAndShortAddress(msg);
  Serial.print("Network ID & Device Address: "); Serial.println(msg);
  DW1000.getPrintableDeviceMode(msg);
  Serial.print("Device mode: "); Serial.println(msg);
  Serial.println("-----------------------------");
  // wait a bit
  delay(10000);
}
-----------------------------------------------------------

At first I thought this incorrect output may be to do with the fact that Arduino Uno pins give out 5v, however as I am getting the Device ID as DECA - model: 1, version: 3, revision: 3, I don't think this is the issue, although I may be wrong.

Once I manage to get this basic connectivity functioning correctly, I aim to use 4 anchors to locate one tag, so it is vital for me to be able to correctly set and retrieve the PAN and Short address of each device.

Please could someone advise me as to why this issue with the PAN and Short Address is occurring and how I may go about solving this?

Thank You!

closed 2018-01-26 by Ritesh23

prisikarm, 2021-09-02:
Hi, I am having the same problem, how did you manage it?
BR, Mike

smartadi, 2022-08-05:
Any update on how this was solved? I am having the same issue directly connecting arduino uno to dwm1000