thotro/arduino-dw1000 #248: Necesito ayuda con mi dwm 1000

Ruben-26, 2017-12-12:
Saludos , tengo algunas dudas que quizá alguno de vosotros me puede solucionar
Quiero hacer un sistema de localización de interiores usando microcontroladores WiFi eso 32  y módulos dwm 1000.
Como esta placa tiene una salida de 3'3 v es necesaria placa adaptadora para el modulo dwm1000 que va a 3'3 también ?
También tengo dudas sobre su conexión con un arduino.He encontrado ejemplos  que dicen:
  Vdda ON a 3'3
Vss a gnd 
IRQ a tierra con una resistencia de 1k y a un un entrada  digital de arruinó
RSTn, SPICLK,SPIMISO,SPIMOSI, SPICsn también a entradas digitales del arruinó
 Es la conexión correcta ? Alguno de esas entradas digitales tiene que ser pwm ?

Rotzbua, 2017-12-13:
English?

closed 2017-12-13 by Rotzbua

Ruben-26, 2017-12-13:
No spanish

Ruben-26, 2017-12-13:
But I can traslate

Ruben-26, 2017-12-13:
Greetings, I have some doubts that maybe one of you can solve me
I want to make an interior location system using WiFi 32 microcontrollers and dwm 1000 modules.
Since this board has an output of 3'3 v, is an adapter plate needed for the dwm1000 module that goes to 3'3 as well?
I also have doubts about his connection with an arduino. I have found examples that say:
Vdda ON at 3'3
Vss to gnd
IRQ to ground with a resistance of 1k and to a digital input of wrecked
RSTn, SPICLK, SPIMISO, SPIMOSI, SPICsn also to digital inputs of the ruined
Is the connection correct? Do any of those digital inputs have to be pwm?