thotro/arduino-dw1000 #249: library copying error

Ruben-26, 2017-12-14:
hello, I downloaded the dwm1000 library and do not copy the examples that it brings, can someone tell me how to fix it?

Ruben-26, 2017-12-14:
I downloaded the library dwm1000 and the examples do not compile
can someone tell me how to fix it?

Ruben-26, 2017-12-14:
this is the error that appears when compiling :

Arduino:1.8.5 (Windows 10), Tarjeta:"ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 115200, None"

In file included from C:\Users\Usuario\Documents\Arduino\libraries\DW1000-0.5.0\examples\BasicConnectivityTest\BasicConnectivityTest.ino:15:0:

C:\Users\Usuario\Documents\Arduino\libraries\DW1000\src/DW1000.h:249:24: fatal error: ../SPI/SPI.h: No such file or directory

compilation terminated.

Se encontraron múltiples librerías para "DW1000.h"
Usado: C:\Users\Usuario\Documents\Arduino\libraries\DW1000
 No usado: C:\Users\Usuario\Documents\Arduino\libraries\DW1000-0.5.0
 No usado: C:\Users\Usuario\Documents\Arduino\libraries\DW1000-0.4.0
 No usado: C:\Program Files (x86)\Arduino\libraries\DW1000-0.5.0
exit status 1
Error compilando para la tarjeta ESP32 Dev Module.

werty37, 2017-12-26:
Hi Ruben,

I can confirm that this library works with ESP8266. ESP32 is untested i believe. Looking at your error it seems like you are missing the SPI.h include in the default Arduino installation. 

`#include <SPI.h>`

Reinstall maybe? 

Cheers
S

Ruben-26, 2017-12-27:
Hi werty37,

I have already managed to avoid that error, apparently I had not cloned the repositories well.
Now I see this error only when I try to load some example of dwm1000 on an esp32 adafruit Feather plate. with arduino plates one or nano if I can or to sibir these examples, although I opt for this result that I do not know if it is the expected one, because I only have one dwm1000.

[image]

[image]

Ruben-26, 2017-12-27:
[image]

werty37, 2017-12-29:
I don't know if someone has been able to interface the DWM1000 with ESP32. As the error suggests, usingInterrupt() is unavailable for ESP devices. You might want to take a look at the DW1000.cpp to understand how ESP8266 is initialized and maybe port it to work with ESP32. 

``` cpp
#ifndef ESP8266
	SPI.usingInterrupt(digitalPinToInterrupt(irq)); // not every board support this, e.g. ESP8266
#endif
```

Here is a thread which explains how ESP32 works
https://github.com/espressif/arduino-esp32/issues/624#issuecomment-353519453

closed 2018-01-25 by Rotzbua