thotro/arduino-dw1000 #254: current consumption in deepSleep mode

vat1, 2018-01-16:
I've measured current consumption from 5 mA on the DWM1000 in sleep time (DW1000.deepSleep();). It's to more then 100 nA mentioned in DW1000 datasheet. Have someone better result or ideas how to decrease it.
thanks in advance!

werty37, 2018-01-17:
DWM1000 datasheet mentions 200nA for deep sleep mode and 550nA for sleep mode. But 5mA in your case seems more than normal.
https://www.decawave.com/sites/default/files/dwm1000-datasheet-v1.6.pdf

It would be easier to understand if you post your schematic.

vat1, 2018-01-17:
Thank you for answer!

I'm using the same datasheet "Figure 9: Example DWM1000 Application Circuit" on page 17 and ESP8266 like controller.
[image]

werty37, 2018-01-17:
The circuit seems fine. Decawave recommends grounding the wakeup pin, if unused though. You might want to check the `void DW1000Class::deepSleep()` method. A bunch of registers are getting set there. 

https://github.com/thotro/arduino-dw1000/blob/master/src/DW1000.cpp

kuek, 2018-01-29:
your schematic does work, but its not power efficient. its probably the LDO.

regards, 
kuek from www.localino.net

seppestas, 2021-01-07:
Not sure if anyone is still interested in this issue, but the problem is likely the LDO. It has a quiescent current of 5mA:
[image]
This means that even if unloaded, the LDO will typically consume 5mA. Also not that this LDO will not be able to handle the transient currents required by DWM1000 in typical operation.

Having the EXTON pin pulled to GND by R3 is also not ideal. This will prevent the DWM1000 from turning of it's internal DC/DC, which is controlled by the EXTON signal. This will add an additional quiescent current (though not nearly as high as that of the LDO).

Make sure to pull down the SPI MISO and SPI CLK pins as @werty37 mentioned.