thotro/arduino-dw1000 #257: Negative Ranges

Ritesh23, 2018-01-26:
Hi All,

I am using the DWM1000 chips, one as an anchor and one as a tag and I am using the DW1000RANGING_TAG/ANCHOR examples, with no alterations to the code.

For some reason I am getting negative values for the range between the two chips, with the negative values decreasing as I move the tag closer to the anchor.

Here is an example of the output I am getting:

[image]

Could anyone help me understand why this issue is occurring?

Thank you.

kuek, 2018-01-29:
calibration is always a good start.

regards,
kuek from www.localino.net

Ritesh23, 2018-01-29:
Hi Kuek, thanks for the response.

I'm a beginner to wireless communications so I am unsure how to calibrate the DWM1000 chip. Do you know of any documentation or code which may be of help for calibration?

Ritesh23, 2018-02-13:
Hi @kuek 

Thanks for your response.

I am new to using these chips so I am not sure how to calibrate them.

Do you have any calibration code or documentation that may help?

Thank you.

Kind Regards,

Ritesh

Rotzbua, 2018-02-18:
Neg. ranges duplicate of #121

closed 2018-02-18 by Rotzbua