thotro/arduino-dw1000 #259: General questions about the DW1000 module

momoshiro94, 2018-01-31:
Hi,

My apologies for my english.

I am planning to buy two or three DWM1000 modules and I have a few questions before buying it.
For each question , the method I want to use is a two way ranging. I would like to put one on a keyfob as a tag, and one in the wall as an anchor.

1) Is the DWM1000 still dependant of the Arduino Mini Pro after we put the tag code inside the module, or will it be able to work with a button cell only?

2) I saw in the datasheet that if I use a button cell as a power supply, it is not recommended to make the module work for too long. Is it possible to put the tag into deep sleep or sleep and only wake it up when it receives a pulse from the anchor?

3) I would like to limit the range detection of the module to 1 meter at the maximum. What I mean is if the tag is more than 1 meter from the anchor for example, the anchor will not take it in count. Is that possible? I saw in the issues that it has to do with the antenna calibration by decreasing the TX power I think.

4) As the keyfob has a lot of chance to be kept close to a body proximity, there would be a dB attenuation somehow because it can be a NLOS case. I would like to know if somebody had tested how much inaccuracy we got if a human body is close to a tag but there is still a LOS connection and when a human body is between the tag and anchor (centimeter or meter error) at 1 meter range?

5) I have a few Arduino Pro mini 5V, if I understand well, I will need to have two sort of PCB, one for the DWM1000 module, and one for the Arduino adapter board ( 5 to 3.3V). I also noticed that after I read a few issues, some users could not be able to make it work with the Arduino mini pro 3.3V due to its low current in the I/O pins, but some of them could make it work, what is the reason?

My aim is to create something similar to a RFID system without the cost involved and a proper range detection.

Thank you for your help in advance.

werty37, 2018-02-01:
> For each question , the method I want to use is a two way ranging. I would like to put one on a keyfob as a tag, and one in the wall as an anchor.

I am assuming you are looking at proximity sensing application and not trilateration / indoor positioning. I would suggest you take a look at bluetooth based proximity sensing as well if you do not require high accuracy. 

> 1. Is the DWM1000 still dependent of the Arduino Mini Pro after we put the tag code inside the module, or will it be able to work with a button cell only?

The host microprocessor, Arduino in you case is still required. The datasheet mentions 64mA for receive mode. In real-world applications it could be more. I am highly skeptical of the DWM1000 working with just a button cell.  The button cell batteries are designed for constant drain of 0.5C or lesser based on their chemistry. Here is a read-up about coin cell. 

https://electronics.stackexchange.com/questions/234901/lithium-coin-cell-cr2032-battery-specifications

> 2. I saw in the datasheet that if I use a button cell as a power supply, it is not recommended to make the module work for too long. Is it possible to put the tag into deep sleep or sleep and only wake it up when it receives a pulse from the anchor?

It is possible to setup the DWMs in Low-Power Sniff mode and Low-Power Listening mode but i am not sure if a button cell can support the burst capacities the DWM would require. It is  certainly possible with a LIPO cell. 

http://thetoolchain.com/mirror/dw1000/dw1000_user_manual_v2.05.pdf#%5B%7B%22num%22%3A76%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C51%2C169%2C0%5D

> 3. I would like to limit the range detection of the module to 1 meter at the maximum. What I mean is if the tag is more than 1 meter from the anchor for example, the anchor will not take it in [...]

You can ignore the ranging requests based on their signal strengths but you might have to do some amount of testing. Thought it is a matter of speculation if there would be even be a considerable signal strength difference say at 0.98m and 1.1m of the tag from the anchor. 

> 4. As the keyfob has a lot of chance to be kept close to a body proximity, there would be a dB attenuation somehow because it can be a NLOS case. I would like to know if somebody had tested how much [...]

There should definitely be attenuation but i have not done any sort of testing on this.  Maybe other can share their knowledge on this topic. 

> 5. I have a few Arduino Pro mini 5V, if I understand well, I will need to have two sort of PCB, one for the DWM1000 module, and one for the Arduino adapter board ( 5 to 3.3V). I also noticed that [...]

The Anchor and Tag can have the same schematic and circuit. You will need to power the DWM1000 separately from the Low-Dropout Regulator (LDO) since the Arduino is not capable of providing power to the DWMs.  You can use 5V Arduinos but you will need Logic Level Conversion since the DWMs are 3.3V devices. You can use Wayne Board though it requires component sourcing and some hacks to get it to work. 

https://www.oshpark.com/shared_projects/2qZiePck
https://sites.google.com/site/wayneholder/uwb-ranging-with-the-decawave-dwm1000---part-ii
https://github.com/thotro/arduino-dw1000/wiki/Testbed-and-Adapter-board#wayne-board
https://github.com/thotro/arduino-dw1000/wiki/Testbed-and-Adapter-board

Or you can use pre-built ones like LPS Minis or Localinos, if you are in a hurry. 

> My aim is to create something similar to a RFID system without the cost involved and a proper range detection.

In my opinion, DWMs will be much more expensive than creating your own RFID based system.

momoshiro94, 2018-02-02:
Hi, 

Thank you for your help.
I already tried to use the BLE 4.0 HM-10 module, but when I put my hand between to cover the LOS path at around 15 cm from the master, the accuracy is not stable anymore, it goes from 15cm to 3-5 meters, it's too much for my project.

asheeshraikwar, 2019-06-18:
Hi everyone , how to get tag and anchor 64 bit  fix address in my dw1000 ic with arduino pro mini libraries??