thotro/arduino-dw1000 #264: Ranging accuracy issues with multiple anchors and 1 tag

Ritesh23, 2018-02-17:
Hi,

I am using the DWM1000 modules (3 anchors and 1 tag), all connected to Arduino Pro Mini 3.3v. I am using the DW1000Ranging_ANCHOR/TAG examples to retrieve ranges for the distances between the tag and the 3 anchors. 

I am able to obtain 3 ranges however I am experiencing large inaccuracies in range values. For example if only Anchor1 is ranging with the tag, the accuracy is not too bad (some negative values when in close proximity), however if Anchor 2 is added, the ranging accuracy for Anchor 2 and the tag is not very good and if Anchor 3 is added, this range is even more inaccurate.

I have changed the MAX_DEVICES variable to a value of 5 in an attempt to resolve the issue however it has not solved the problem.

Does anyone have this same issue and has managed to resolve it?

Thank you.

kuek, 2018-02-17:
you probably have the same name for the tags? 

kuek from www.localino.net

Rotzbua, 2018-02-18:
There are thousands of this issues, please use search.

closed 2018-02-18 by Rotzbua