thotro/arduino-dw1000 #270: Anchor is not always reconnecting properly

gautierhattenberger, 2018-03-22:
### Step 1: What do you want?
Report a possible issue in the ranging library

### Step 2: Describe your environment
 * Arduino IDE version or Toolchain: 1.8.1
 * dw1000 Library version: latest master https://github.com/thotro/arduino-dw1000/commit/e20c3b91c3d6cee0ef848aa41e72c942cb0ac7df
 * Arduino device: pro mini (tag) and nano (anchor)
 * Used adapter board: custom based on wiki instructions
  
### Step 3: Describe the problem
The experiment is just to test the distance between an anchor and a tag. If the tag goes too far, the device is properly removed, and when coming back, connection is good again.
But when moving slowly at the edge of the reception range, sometimes the connection is lost and never comes back after that until the anchor is restarted (restarting the tag has no effect)

Find bellow the output of the last messages from the TAG:
```
from: 1	 Range: 24.16 m	 RX power: -94.62 dBm
from: 1	 Range: 24.17 m	 RX power: -93.75 dBm
from: 1	 Range: 23.97 m	 RX power: -95.49 dBm
from: 1	 Range: 23.89 m	 RX power: -94.63 dBm
from: 1	 Range: 24.24 m	 RX power: -96.11 dBm
from: 1	 Range: 24.57 m	 RX power: -97.00 dBm
from: 1	 Range: 24.55 m	 RX power: -99.11 dBm
delete inactive device: 1
ranging init; 1 device added ! ->  short:1
from: 1	 Range: 25.36 m	 RX power: -98.98 dBm
from: 1	 Range: 25.19 m	 RX power: -98.31 dBm
from: 1	 Range: 25.22 m	 RX power: -98.23 dBm
from: 1	 Range: 25.14 m	 RX power: -98.76 dBm
from: 1	 Range: 25.26 m	 RX power: -99.52 dBm
delete inactive device: 1
ranging init; 1 device added ! ->  short:1
from: 1	 Range: 25.28 m	 RX power: -98.43 dBm
from: 1	 Range: 25.32 m	 RX power: -102.25 dBm
from: 1	 Range: 25.17 m	 RX power: -98.00 dBm
from: 1	 Range: 25.05 m	 RX power: -98.74 dBm
from: 1	 Range: 25.23 m	 RX power: -98.24 dBm
from: 1	 Range: 25.40 m	 RX power: -98.17 dBm
from: 1	 Range: 25.34 m	 RX power: -98.37 dBm
from: 1	 Range: 25.36 m	 RX power: -98.34 dBm
from: 1	 Range: 25.26 m	 RX power: -99.13 dBm
delete inactive device: 1
ranging init; 1 device added ! ->  short:1
delete inactive device: 1
```

and from the ANCHOR:
```
from: 0	 Range: 24.16 m	 RX power: -94.62 dBm
from: 0	 Range: 24.17 m	 RX power: -93.75 dBm
from: 0	 Range: 23.97 m	 RX power: -95.49 dBm
from: 0	 Range: 23.89 m	 RX power: -94.63 dBm
from: 0	 Range: 24.24 m	 RX power: -96.11 dBm
from: 0	 Range: 24.57 m	 RX power: -97.00 dBm
from: 0	 Range: 24.55 m	 RX power: -99.11 dBm
delete inactive device: 0
blink; 1 device added ! ->  short:0
from: 0	 Range: 25.36 m	 RX power: -98.98 dBm
from: 0	 Range: 25.29 m	 RX power: -97.93 dBm
from: 0	 Range: 25.19 m	 RX power: -98.31 dBm
from: 0	 Range: 25.22 m	 RX power: -98.23 dBm
from: 0	 Range: 25.14 m	 RX power: -98.76 dBm
from: 0	 Range: 25.42 m	 RX power: -98.47 dBm
from: 0	 Range: 25.26 m	 RX power: -99.52 dBm
delete inactive device: 0
blink; 1 device added ! ->  short:0
from: 0	 Range: 25.28 m	 RX power: -98.43 dBm
from: 0	 Range: 25.32 m	 RX power: -102.25 dBm
from: 0	 Range: 25.17 m	 RX power: -98.00 dBm
from: 0	 Range: 25.05 m	 RX power: -98.74 dBm
from: 0	 Range: 25.23 m	 RX power: -98.24 dBm
from: 0	 Range: 25.40 m	 RX power: -98.17 dBm
from: 0	 Range: 25.34 m	 RX power: -98.37 dBm
from: 0	 Range: 25.36 m	 RX power: -98.34 dBm
from: 0	 Range: 25.26 m	 RX power: -99.13 dBm
from: 0	 Range: 25.32 m	 RX power: -101.32 dBm
delete inactive device: 0
blink; 1 device added ! ->  short:0
delete inactive device: 0
```

As you can see, the only difference between to cases that worked and the last one that didn't is that one range have been measured by the ANCHOR (`from: 0	 Range: 25.32 m	 RX power: -101.32 dBm`) but not by the TAG. After that, the blink from the tag is received but no more measurements even when coming back close to the anchor.

Does anyone have an idea what could be the issue ? Can it be in the ranging logic stuck in a state waiting for a message that will never arrive ? Is there any error message that I could print to have a better idea of the current device state ?

Foralux, 2018-03-27:
Hi ,

You are not alone. I've finally managed to get my setup up and running but I've also been hit with the same issue. I setup 4 anchors in a 10x10m square . I turn on the tag , and it takes about not even 10 seconds before the Tag or Anchor stops receiving data. As you said, the Leds stop blinking. Restarting the tag doesnt seem to work. I have to go around and restart all anchors and even then it's a bit hit and miss. 

I am using STM32's

Foralux, 2018-03-28:
Followup - I've implemented a fix from the user here https://github.com/thotro/arduino-dw1000/pull/236/commits/7be23f0981645a85eba913a39097f87b3e456a34 Into the latest commit. Seems like I'm not experiencing the issue anymore. But I am still going to do some more testing as I might be wrong :P

gautierhattenberger, 2018-03-29:
@Foralux Thanks for the hint. It did solve my issue. I just made a new pull request as the initial one (#236) was missing one test.

Rotzbua, 2018-04-07:
fixed by 65544425521717da3c45db211cfb536aaf262250

closed 2018-04-07 by Rotzbua