thotro/arduino-dw1000 #273: Sample rate fixed a 10Hz?

tistructor, 2018-06-21:
Hello everybody,
I'm doing experiments using the sample files for reading the distance between a TAG device and an ANCHOR. (DW1000Ranging_TAG.ino, DW1000Ranging_ANCHOR.ino) the system works but I get a reading frequency of 10 HZ I tried to have a look at the files library to determine if a possible constant determined this operation.
I did not find anything.

I did experiments with the other two examples where to determine the distance between devices do not use the DW1000Ranging.h file and in this case I get 80Hz but the frame rate is not stable and sometimes it stops and then start again.
I would need a stable frame rate at around 40Hz. 10Hz are few.

I thank those who will have some suggestions to give me.

Fabio

Rotzbua, 2018-10-01:
@tistructor Did you found a solution? Can I close this?

closed 2018-10-10 by Rotzbua

mehtivakili, 2024-12-15:
Hi
This is what I exactly looking for