thotro/arduino-dw1000 #276: Speed adjustment Mini Pro 3.3V v Mini Pro 5V

orbitcoms, 2018-07-11:
Hi,
I found this site when looking for Arduino Compatible code for DWM1000 modules.

I understand the code in the repo is based on Mini-Pro 3.3V, 8MHz processor, is this correct?

If the processor was changed to the 5V version running at 16MHz, what parts of the code would need to be modified?

Finally, Is the code on this page the latest available?

Thanks in Advance
Dave

Rotzbua, 2018-07-20:
This library is written for arduino hardware with the arduino framework. I think there is no code dependent on the clock speed. I guess it would have an impact on your measurements, eg. as a constant time offset.

Sonic0, 2018-07-20:
same thing that Rotzbua says. Make sure the pins are the same.
I think that with the 5V-16MHz version of Arduino you can obtain a more accurate measure. more frequency clock is very useful when your tag is in motion; Arduino can read more interrupt from IRQ.

closed 2018-07-27 by Rotzbua