thotro/arduino-dw1000 #290: WARNING! Software license is incompatible with Arduino SPI library license

an4cr0n, 2018-09-07:
SPI library code is licensed under GPL-2.0 and this is Apache 2.0
As stated in https://www.apache.org/licenses/GPL-compatibility.html apache is never been considered gpl-2.0 compatible and the main problem is if you use code under GPL-2.0 you must release your code under GPL-2.0 or more. This repo should then be converted to GPL-2.0 or GPL-3.

Rotzbua, 2018-09-07:
Which SPI library is published under GPL-2.0? The AVR SPI library is under LGPL-2.1.

an4cr0n, 2018-09-07:
Sorry, i missed that 'lesser' in the file.

Rotzbua, 2018-09-07:
Ok. LGPL and Apache are compatible or did I missed something?

Maybe the examples should be published under a more open license than Apache like CC0. Do you have more experience with license @an4cr0n ?

an4cr0n, 2018-09-08:
@Rotzbua i don't have much experience, but apache is usually permissive and i think the example should remain as they are (as long as you don't specify different actions regarding liability and such.) .

While searching compatibility with LGPL i stumbled upon this link: https://www.apache.org/legal/resolved.html
In section:"WHICH LICENSES MAY NOT BE INCLUDED WITHIN APACHE PRODUCTS?" LGPL is one of them but i can't understand if using a LGPL library is considered 'including' in Apache products.
I can't find anything on the web in regard to this. Should we reopen this and change the title to something like 'Software license may be incompatible with Arduino SPI library license' ?

an4cr0n, 2018-09-08:
From https://issues.apache.org/jira/browse/LEGAL-192 it seems like apache2 does allow LGPL only in the case of 'optional' features and if you provide a way to add or remove the optional features.
As SPI is used as a main feature this would mean that apache 2.0 could be inappropriate, but i'm no lawyer nor an expert, i'll reopen this with the title mentioned above and hope someone who knows his thing can help use, thanks for your time @Rotzbua

Rotzbua, 2018-10-11:
> seems like apache2 does allow LGPL 

They do not allow LGPL because it may "overwrite" the apache license.

----

I added a list of dependencies and license hints.

----

I did some reading:

All arduino binaries are static linked. If LGPL libraries are static linked, the binary is a derivative of the LGPL libraries. This means the binary has to be distributed under LGPL license.

In fact this means that every program you create with parts of the arduino framework has to be distributed under LGPL.

In conclusion it is not recommended to compile code which contains secret intellectual property with LGPL libraries and distribute it to customers. Because the customers can ask for the source code according to the LGPL license. (This was birth of OpenWRT because tp-link ignored GPL and was sued)

If you use the binary only by your self or you have no problem with sharing your code with your customers, then #yolo .

My personal conclusion: the users of this library have to think about how and what code under which license they use. I can't protect persons from license violations.

BTW: at my earlier work any usage of any GPL code in application was forbidden. LGPL was only allowed with dynamic linkage and only with confirmation of the team.

closed 2018-10-11 by Rotzbua