thotro/arduino-dw1000 #296: Noisy transmission

diegocaples, 2019-03-08:
Hello, I just got two decawave dw1000's working with 2 Arduino Uno's (without using the adapter board), and they can successfully transmit and receive data. 
Unfortunately, the transmissions contain a lot of noise. Sometimes I will receive a clean transmission. but usually the messages are garbled. The two arduinos are running the
basic receiver and basic sender sketch. Here some sample  received data is: 

### DW1000-arduino-receiver-test ###
DW1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: EUI: 0:0:0:0:255, OUI: 255:255:255
Network ID & Device Address: PAN: 10, Short Address: 6
Device mode: Data rate: 110 kb/s, PRF: 16 MHz, Preamble: 1024 symbols (code #4), Channel: #5
Received message ... #1 
Data is ... Hello DW1000, it's #43‰ü·×Çàªƒr='
EC¢é=£‘,
Received message ... #2
Data is ... ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
Received message ... #3
Data is ... ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
Received message ... #4
Data is ... Hello DW1000, it's #461‚·×Çàªƒr='

It is very rare that I get a clean transmission. Here one is:
Received message ... #8
Data is ... Hello DW1000, it's #50 

There a lot of messages like
ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
with different numbers of y's.

Is there anything I am doing wrong? Thanks for help in advance!

diegocaples, 2019-03-08:
I am using the Arduino 3.3 output to power the board

curlyz, 2019-03-13:
please use external power supply , add a bunch of capacitor and it will work

diegocaples, 2019-03-13:
Ok, Thanks! But where should I put the capacitors, and what kind of capacitor should I use?

curlyz, 2019-03-13:
On the 3.3V line of course , use an external AMS1117 to convert 5V to 3.3V
rail only for the DWM1000 , put some large eletrolyc cap , a ~ 10uF ceramic
and ~ 0.1uF too.

diegocaples, 2019-03-15:
Do all of the capacitors go between the 3.3V rails?

closed 2019-04-08 by diegocaples