thotro/arduino-dw1000 #297: Apparent overflow issue

UserIoana, 2019-06-06:
Hi! I would like some help with the DW1000Ranging_TAG/ ANCHOR examples. 
I have two dwm1000 modules, one connected to an arduino pro mini (the 3.3V one) (the tag) and one connected to an Arduino due (the anchor). The examples BasicConectivity, BasicSender and BasicReceiver work fine, but when I try to get the range between the two, using the DW1000Ranging_ANCHOR/TAG examples, I get results like this :
[image]
I understood that this might be an overflow error but I have no idea what could cause it, considering the fact that the sketches worked fine for other people. I would appreciate some advice, thank you!

wahajmurtaza, 2019-07-25:
Have you changed the channel? And for what distance you are receiving these values?

kihoonSon, 2022-01-27:
Hello!
I'm having the same problem as you.
Did you solve this problem?