thotro/arduino-dw1000 #300: 2 TAG AND 4 ANCHOR RANGE NOT RECEIVING FROM BOTH THE TAG

asheeshraikwar, 2019-07-08:
Hi, I am using dw1000 ic with Arduino pro mini.installing 2 tags and 4 anchors give the following disrupted range.
[image]
 4 anchors with 1 tag are working fine. as shown below
[image]

need any help on 2 tags and 4 anchors issue

wahajmurtaza, 2019-07-09:
I got the same problem, I just ignored the negative readings and the values beyond my range of work,
It caused some delay in  values, but did good for my application.

nagendra092918, 2019-09-10:
@asheeshraikwar can u please provide the code of 4 anchors and 1 tag. i tried a lot but i did not please help me
mail id: nagendra092918@gmail.com
i hope u will help me
thank you

wahajmurtaza, 2019-09-10:
@nagendra092918  just use dwm1000Ranging _ANCHOR on 4 arduinos and dwm1000Ranging_TAG from example code. No big deal.

nagendra092918, 2019-09-10:
@wahajmurtaza tq for u reply if i connect 3 anchros at a time i am not getting distances properly i got negative  values and i need all measured distance in tag . tag is connected to laptop i need find the position of the tag . if u have programing please help

                    thank you

wahajmurtaza, 2019-09-10:
Check my repository of Indoor Positioning System, I have made custom library for me. You can skip the node mcu part and just connect tag serially to view data. https://github.com/wahajmurtaza/Indoor-Positioning-and-Navigation-using-Reinforcement-Learning
It was programmed for 3 anchors but I am sure you can manage.

wahajmurtaza, 2019-09-10:
@nagendra092918  You must delete the library and download the one which is mentioned in the link.
I was getting accurate results using 3 anchors.

nagendra092918, 2019-09-10:
tq bro