thotro/arduino-dw1000 #301: DW1000Ranging_TAG No feedback

583345824, 2019-07-18:
[image]
I used two sets of DWM1000+minipro
They can run BasicReceiver/Sender
But when I uploaded A and B, TAG didn't seem to receive the message?

wahajmurtaza, 2019-07-21:
Send circuit diagram that you are using.

CFH00, 2020-02-15:
> [image]
> I used two sets of DWM1000+minipro
> They can run BasicReceiver/Sender
> But when I uploaded A and B, TAG didn't seem to receive the message?

> [image]
> I used two sets of DWM1000+minipro
> They can run BasicReceiver/Sender
> But when I uploaded A an

> [image]
> I used two sets of DWM1000+minipro
> They can run BasicReceiver/Sender
> But when I uploaded A and B, TAG didn't seem to receive the message?

Hi, Friend
Have you sloved it out?

DAGANGYE, 2020-07-30:
I have the same problem, the following is the circuit diagram
[image]

wahajmurtaza, 2020-08-02:
Have you tried changing the channel and increasing the power?

CFH00, 2020-08-03:
Thanks for reply. I have sorted it out. How is your project going?

University of Huddersfield inspiring global professionals.
[http://marketing.hud.ac.uk/_HOSTED/EmailSig2014/EmailSigFooterMarch2019.jpg]

This transmission is confidential and may be legally privileged. If you receive it in error, please notify us immediately by e-mail and remove it from your system. If the content of this e-mail does not relate to the business of the University of Huddersfield, then we do not endorse it and will accept no liability.

Yasirsmb, 2020-08-25:
> Thanks for reply. I have sorted it out. How is your project going?
> […](#)

can you help me to solve this problem

CFH00, 2020-08-25:
> > Thanks for reply. I have sorted it out. How is your project going?
> > […](#)
> 
> can you help me to solve this problem

so what is your problem

socconlc, 2020-10-27:
> Thanks for reply. I have sorted it out. How is your project going?
> […](#)

I have the same problems with you. Could you help me to solve that?

Savage33222233, 2021-04-28:
How solve this problem?

Savage33222233, 2021-04-28:
> Спасибо за ответ. Я разобрался. Как продвигается ваш проект?
> […](#)

How solve this problem?

wahajmurtaza, 2021-05-01:
Try the forked repository of dw1000 at my profile or I have a pull request here,
There you can select channel. I got best connectivity at channel 2.
Also make sure you are providing sufficient power to your board.

Savage33222233, 2021-05-06:
> Try the forked repository of dw1000 at my profile or I have a pull request here #303 ,
> There you can select channel. I got best connectivity at channel 2.
> Also make sure you are providing sufficient power to your board.

I installed your sketch with your library, but it's doesn't work. About power, I am using a DC module MB-102 for the breadboard. What else could be the problem?