thotro/arduino-dw1000 #304: DWM1000 robotics

FXUDarkNova, 2019-08-30:
I am looking to use 3 anchors and 1 tag to make a robot follow me I have got the 3 anchors setup and two sending the information to one  main anchor. I am know stuck on where to go from here using the data I am collecting could someone point me in the right direction?

nxdefiant, 2019-08-30:
[I made my robot follow a tag using two anchors](https://hackaday.io/project/25406-wild-thumper-based-ros-robot/log/152597-follow-me-part-4). Maybe that helps?

FXUDarkNova, 2019-09-03:
Thank you I shall take a look would you recommend only using two anchors instead of 3?

nxdefiant, 2019-09-03:
I guess the math is a bit simpler when using only two anchors, so I would say yes. You can always add the third anchor later. Be aware that using only two anchors you get two solutions for the equation. One in front of the robot, one behind.

FXUDarkNova, 2019-09-03:
Slightly confused looking at your code and the diagram is using two anchors not used to determine left and right? Not front and back?

rbrtwtrs, 2019-09-03:
It is the intersection of two spheres of range so two possible solutions.
With 3 the ambiguity is resolved.

nxdefiant, 2019-09-03:
Indeed, I have the two assumptions that the tracking starts in front of the robot and that the robots does everything successfully for the tracked object to stay in front.

FXUDarkNova, 2019-09-04:
Ok makes sense just one question when you do t = dist_right**2 - a_r**2 is that working out turn?

nxdefiant, 2019-09-04:
Sorry, I did not understood your question (not a native speaker here)

FXUDarkNova, 2019-09-04:
It’s ok iv worked that bit out i am confused as x cords seems to be working as Y cords very confusing

nxdefiant, 2019-09-04:
Yeah, for some reason I decided to use two different coordinate systems. That is why I turn it later with 
`x, y = (y, -x)`

FXUDarkNova, 2019-09-04:
I have that bit in my I’m doing it in c language and all I get is x y having same values just y is negative

nxdefiant, 2019-09-04:
I recommend to visualize the values, it helps a lot. I have that "VISUALIZE" in my script for that.

FXUDarkNova, 2019-09-04:
Ok thank you I will give it ago

nilesh999, 2022-12-01:
Hello, I have been working dwm1000 module. I would like to know about tag 1 and anchor3. I just want to get collection of positions and distance between tag and anchors. I have make condition with distance along with heading to get navigation of motor. For kind information I want know motor movement as per distance and position of anchors. I am looking dwm1000 source code for Robot. 
Could you please share source code.

nilesh999, 2022-12-01:
I am working with Arduino and dwm1000 for tag and anchors