thotro/arduino-dw1000 #306: can any one help me 3 anchors and 1 tag program. what are changes i need to do in program

nagendra092918, 2019-09-10:
### Step 1: What do you want?

* Report an issue.
* Improve code (and need help).
* Suggest a new feature.
* Present a new hardware adapter board or wiring.
* Want to report my experience or measurement. (Please also add them to the wiki!)

### Step 2: Describe your environment

 * Arduino IDE version or Toolchain: _____
 * dw1000 Library version: _____
 * Arduino device: _____
 * Used adapter board: _____
  
### Step 3: Describe the problem

#### Affected file(s) or example(s):
  
  * Filename 1
  * Filename 2
  
#### Steps to reproduce:

  1. _____
  2. _____
  3. _____
  
#### Observed Results:

  * What happened?  This could be a description, log output, etc.
  
#### Expected Results:

  * What did you expect to happen?
  
#### Relevant Code:

```c++
  // TODO(you): code here to reproduce the problem
```

wahajmurtaza, 2019-09-10:
Nothing, Just upload 3 dwm1000Ranging_Anchor and 1 dwm1000Ranging_Tag. And If you want them to not be random address all time, just use different address and add new parameter 'false' in the start as function.
You can view my repository at https://github.com/wahajmurtaza/Indoor-Positioning-and-Navigation-using-Reinforcement-Learning

nagendra092918, 2019-09-10:
tq bro i will test. if i stuck any where i will ask. please help

Sonic0, 2019-09-10:
Also, see other projects in Github. 
A good refactor of thotro's library and web based RTLS infrastructure is [F-Army](https://github.com/F-Army)

nagendra092918, 2019-09-10:
tq somuch bro

nagendra092918, 2019-09-10:
bro i am getting this is the output printing on serial monitor i think so this is wrong
[image]

wahajmurtaza, 2019-09-10:
Did you change the baudrate and print format? and Did you replace the library?
Because there was no code for device address printing and "final_string".
Try 9600 baudrate, I increased the baudrate and my device kept restarting.

wahajmurtaza, 2019-09-10:
Discuss it on my repository, Its not an issue here.

Fuhu-che, 2019-12-11:
> Did you change the baudrate and print format? and Did you replace the library?
> Because there was no code for device address printing and "final_string".
> Try 9600 baudrate, I increased the baudrate and my device kept restarting.

Hi bro, I used your code but shows "no such file or directory". can I ask is there anything wrong?