thotro/arduino-dw1000 #312: Program the DWM 1000

Fuhu-che, 2019-11-25:
HI all,
I am working on indoor localization project now. I built the PCB for DWM1000 and plan to connect with Arduino pro mini or nano. but before connection, can anybody tell me do I need to program the DWM 1000 first or just connect them and using arduino to program it. Thanks a lot!

snyderth, 2019-12-16:
You do not "Program" the DWM 1000 chip in the sense of loading a program into memory. Instead, you configure it via SPI on startup. Connect the SPI pins to the proper connections on the DWM 1000 and this will allow you to configure it. The [DWM 1000 user manual](https://www.decawave.com/sites/default/files/resources/dw1000_user_manual_2.11.pdf) is a great resource, and @thotro used the same naming conventions for the configuration parameters as are listed in the manual. 

Just a reminder that the DWM1000 runs on 3.3v logic.

Fuhu-che, 2019-12-17:
> You do not "Program" the DWM 1000 chip in the sense of loading a program into memory. Instead, you configure it via SPI on startup. Connect the SPI pins to the proper connections on the DWM 1000 and [...]

Thank you for you help and do you mean DW1000 user manual?

snyderth, 2019-12-17:
> Thank you for you help and do you mean DW1000 user manual?

Yes

Fuhu-che, 2019-12-27:
Hi, Bro,
The DW1000 manual is too long and difficult to read it. Any other good suggestions? Thanks

snyderth, 2019-12-29:
It is difficult, but with patience, readable. The interface for the DW1000 chip is SPI. Through the SPI interface, you set registers (32-bit values) based on your necessary configuration. This library has good examples for doing so. I suggest hooking up the Arduino pro mini or nano and starting with one of the examples that does not use the DW1000Ranging class so that you can see the inner workings of the code and what it means to write to registers via SPI. A lot of projects require much patience. Tracing the code will not be easy, but it is very beneficial. It can be boring and tedious, but it pays off, and you will learn a lot about how sensors work.

Fuhu-che, 2019-12-29:
> It is difficult, but with patience, readable. 

Hi Bro,
Thank you so much for your suggestion an encourage me. Can I ask  you which library has good example (do you mean some example on arduino?)and is it possible to leave your email that can make us talking easier? If not convenient, forget it. Thanks.