thotro/arduino-dw1000 #315: hi all. Can you help me . How to add more Tag on system ? Thanks you.

jack-vn, 2020-02-10:
### Step 1: What do you want?

* Report an issue.
* Improve code (and need help).
* Suggest a new feature.
* Present a new hardware adapter board or wiring.
* Want to report my experience or measurement. (Please also add them to the wiki!)

### Step 2: Describe your environment

 * Arduino IDE version or Toolchain: _____
 * dw1000 Library version: _____
 * Arduino device: _____
 * Used adapter board: _____
  
### Step 3: Describe the problem

#### Affected file(s) or example(s):
  
  * Filename 1
  * Filename 2
  
#### Steps to reproduce:

  1. _____
  2. _____
  3. _____
  
#### Observed Results:

  * What happened?  This could be a description, log output, etc.
  
#### Expected Results:

  * What did you expect to happen?
  
#### Relevant Code:

```c++
  // TODO(you): code here to reproduce the problem
```

CFH00, 2020-02-21:
> Step 1: What do you want?
> 
> 
> Report an issue.
> Improve code (and need help).
> Suggest a new feature.
> Present a new hardware adapter board or wiring.
> Want to report my experience or measurement. (Please also add them to the wiki!)
> 
> Step 2: Describe your environment
> 
> Arduino IDE version or Toolchain: _____
> dw1000 Library version: _____
> Arduino device: _____
> Used adapter board: _____
> 
> Step 3: Describe the problem
> Affected file(s) or example(s):
> 
> Filename 1
> Filename 2
> 
> Steps to reproduce:
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> Observed Results:
> 
> What happened?  This could be a description, log output, etc.
> 
> Expected Results:
> 
> What did you expect to happen?
> 
> Relevant Code:
>   // TODO(you): code here to reproduce the problem

Hi Friend,
Have you worked out 1 tag and 1 anchors. I cannot get feedback from tag. can you help me?
[image]

anthony211212, 2024-04-28:
Hi. Could you solve the problem in the library about multiple tags?

pizzo00, 2024-09-11:
**Update for anyone who encounter this problem**
You can find an updated version that fixes this and other problems here: https://github.com/jremington/UWB-Indoor-Localization_Arduino/tree/main/DW1000_library_pizzo00