thotro/arduino-dw1000 #318: I have no problems at all using Arduino Pro Mini (3.3v version).

Yasirsmb, 2020-10-28:
I have no problems at all using Arduino Pro Mini (3.3v version). 
I have attached the Arduino and Windows code that I did use in the demo videos (#62).
[dw1000.zip](https://github.com/thotro/arduino-dw1000/files/130502/dw1000.zip)

_Originally posted by @greymfm in https://github.com/thotro/arduino-dw1000/issues/54#issuecomment-184163980_

I found this code and it works well to find the distance between anchor and tag, I need to know it mapped values on PC window

socconlc, 2020-10-29:
Hello, I've used your code but I can't upload the DW1000Ranging_Tag but the DW1000Ranging_Anchor worked fine
This is the problems:
Arduino: 1.8.9 (Windows 10), Board: "Arduino Pro or Pro Mini, ATmega328P (5V, 16 MHz)"

Sketch uses 22188 bytes (72%) of program storage space. Maximum is 30720 bytes.
Global variables use 1193 bytes (58%) of dynamic memory, leaving 855 bytes for local variables. Maximum is 2048 bytes.
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x9a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x9a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x9a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x9a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x9a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x9a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x9a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x9a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x9a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x9a
Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.
Invalid library found in C:\Users\Long Trinh\Documents\Arduino\libraries\espNowFloodingMeshLibrary-master: Missing 'paragraph' from library
Invalid library found in C:\Users\Long Trinh\Documents\Arduino\libraries\espNowFloodingMeshLibrary-master: Missing 'paragraph' from library

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Could you help me to solve that pls?

Yasirsmb, 2020-10-29:
Your board is not connected or selected wrong in arduino IDE

socconlc, 2020-11-01:
> Your board is not connected or selected wrong in arduino IDE
> […](#)

 Hello again. I've fixed that problems but it's still don't work.
I've received something like this from the Tag:
[image]

Could you help me pls?

socconlc, 2020-11-03:
Never mind I've already fixed it. But the range seems like doesn't right. Any recommended

socconlc, 2020-12-04:
@Yasirsmb Are you still there??
Can I ask you one more question pls.
Your code is working fine for me with 1 or 2 anchors. But when I add the 3rd anchor, it's killed immediately and reconnected and killed again. 
I've searched for the problem but can't find any solution with your code.
Do you know how to fix it?