thotro/arduino-dw1000 #322: Using library with DWM1000

heroescharge, 2021-07-27:
I tried using this library with the DWM1000 module, however the Tag and Anchor programs did not seem to work upon upload: no communication occurred between the tag and anchor. I know for sure that the modules function properly as I am able to run my own test programs. Are there any modifications needed to this code so that it will work with the DWM1000, given that the tag/anchor programs are quite old?

viaSeunghyun, 2021-08-02:
No, This Library is work.
I tested DWM1000 and Atmega328p.

Are you test DW1000Ranging_ANCHOR to DW1000Ranging_Tag or RangingAnchor to RangingTag?

sharifsagar80, 2021-09-08:
I am also using trying to use the library but not working. I am using Arduino Uno and DWS1000 Arduino Shield.

Chrisrdouglas, 2021-10-07:
Oh dang, so you didn't get it to work with the arduino uno at all?

I've set the IRQ pin to be on pin 2 and the reset to be on pin 3

ZakariaPZ, 2021-12-15:
@Chrisrdouglas Were you able to fix yours, and get it to work on the Uno?

Chrisrdouglas, 2021-12-15:
No, I got super busy with work and haven't had the time to work on it in a while :/

Are you having similar problems? And did you try calibrating the antennas?

ZakariaPZ, 2021-12-15:
@Chrisrdouglas I haven't actually started my project yet, but I've gotten some Unos and DWM1000s on order. I'm planning on following a certain tutorial setup, so I'm hoping it works out. Do you know how to calibrate the antennas?

sharifsagar80, 2021-12-15:
@ZakariaPZ  Please refer to the DW1000 user manual v2. They give you a comprehensive description of the antenna delay calibration part.

ZakariaPZ, 2021-12-15:
@sharifsagar80 Did you use this DW1000 library with the Nucleo?

sharifsagar80, 2021-12-17:
Yes. I have used the DW1000 library with Nucleo.

sharifsagar80, 2021-12-17:
> @sharifsagar80 Did you use this DW1000 library with the Nucleo?

Yes. I have used the DW1000 library with Nucleo