thotro/arduino-dw1000 #326: DWM1000 NodeMCU Esp8266 3 anchors and 1 Tag

Aria98Esc, 2021-12-09:
### Step 1: What do you want?

* Report an issue.
* Improve code (and need help).
* Suggest a new feature.
* Present a new hardware adapter board or wiring.
* Want to report my experience or measurement. (Please also add them to the wiki!)

### Step 2: Describe your environment

 * Arduino IDE version or Toolchain: _____
 * dw1000 Library version: _____
 * Arduino device: _____
 * Used adapter board: _____
  
### Step 3: Describe the problem

#### Affected file(s) or example(s):
  
  * Filename 1
  * Filename 2
  
#### Steps to reproduce:

  1. _____
  2. _____
  3. _____
  
#### Observed Results:

  * What happened?  This could be a description, log output, etc.
  
#### Expected Results:

  * What did you expect to happen?
  
#### Relevant Code:

```c++
  // TODO(you): code here to reproduce the problem
```
Hello, I have 4 DWM1000 modules. I use 3 of them as anchors and 1 as a tag. I have managed to communicate 1 anchor to the tag, but the other 2 dont work properly. This 2 can only transmit (I tested them with the basic receiver and sender and they can only transmit). So, I tried with the ranging anchor and tag codes and there is no answer from those modules. I also tested them with the basic connectivity and then they do work properly. All the data is shown there. I dont know what Im doing wrong with the ranging anchor code in this modules that dont work. Could you help me with that? I am using a nodeMCU ESP8266 with the DWM1000 and I dont have this modules in a PCB. I just welded wires to then pins. Im using the last version of Arduino and Thotro's library. 
[image]
[image]
Connections:
Esp8266 -> Dwm1000
D1-> RST
D2->IRQ
3V -> 3V
GND -> GND
D5->SCK
D6->MISO
D7->MOSI
D8->CS

Aria98Esc, 2021-12-10:
[image]

ZakariaPZ, 2021-12-15:
Hi, I'm also working on a project with the DWM1000. Did you figure this out?

az-z, 2021-12-15:
I dont think this lib is functional enough past simple ping-pong between 2
devices.