thotro/arduino-dw1000 #335: Arduino Pro mini & DWM1000

itfeels, 2023-06-12:
Basic Sender and Receiver seems working on my Arduino.

[image]

But, When I upload DW1000Ranging_ANCHOR and TAG,

[image]

Distance measurement is messed up like above. It's only 70~80cm between 2 DWM1000.

I'm using DW1000 example code

[image]

Farzad1989, 2023-06-17:
Hi, Is it necessary to use a host processor permanently for tags after the initialization?

amitx64, 2023-06-19:
when I used DW1000 with Arduino Uno it did not work. just, shows ANCHOR and TAG addresses.

amitx64, 2023-06-19:
Did you change in Arduino uno code. So, it can work for you. If it is than please share.